speeduino/speeduino #730: Intermittent PWM Idle at 100% duty cycle

DeionSi, 2021-11-27:
The PWM Idle is intermittent at 100% duty cycle. Example output is below.
[image]

A workaround is to change
SET_COMPARE(IDLE_COMPARE, IDLE_COUNTER + (idle_pwm_max_count - idle_pwm_cur_value) );
to
SET_COMPARE(IDLE_COMPARE, IDLE_COUNTER + (idle_pwm_max_count - idle_pwm_cur_value) + 1 );

The correct fix is probably to do something like PWM fan is doing. At 100% and 0% duty cycle the PWM is disabled and the pin is just set to on or off.

shiznit304, 2022-01-07:
this can probably be closed, looks like it was fixed

noisymime, 2022-01-07:
Ahhh I hadn't realised there was an issue open for this. Yes it was resolved in 464933b38f37391246c19bce5679e817dbb1e0ad

closed 2022-01-07 by noisymime