speeduino/speeduino #737: Re bin Y Axis only allowing even numbers.

AshtonRey, 2021-12-05:
When I try to adjust my Y axis for load, speeduino automatically changes my inputted values to only even numbers. But for ALPHA-N I’d like to use the odd numbers for a better tuning of my spark, fuel, and AFR maps.  Please see image below of what I am referring too. Thanks!!

[image]

LPG2CV, 2021-12-05:
it's done that way to save memory space
the firmware interpolates the values, so it shouldn't be an issue.  so if you enter say 2,4,6 in the table, with corresponding VE of 28,30,36, then if tps is 3, then VE would be 29. 5 tps would return 33.

jonbills, 2021-12-18:
I'd like to use odd values too.
when VE at 0% TPS is 30 and VE at 2% TPS is 60, interpolation of a value for 1% TPS is a wild guess

VitorBoss, 2021-12-18:
#608 adresses that and allow you to use 0.5% steps

jonbills, 2021-12-18:
> #608 adresses that and allow you to use 0.5% steps

ah, I'd thought that was only for AE, but you're right - looks great, will it be merged soon? or should I install and test the branch?

noisymime, 2022-01-05:
This has been resolved in #608

closed 2022-01-05 by noisymime