speeduino/speeduino #740: There is more than 1 ignition for 1 trigger input for 1,2,5 cylinder using basic distributor and single coil output.

JuanNavarroS, 2021-12-10:
In basic distributor mode + single coil output, for 1, 2, 5 cyl, there is a OK spark in synchro with the trigger event (equivalent to breaker point lobe), but one extra spark KO in the middle between 2 trigger events.

jonbills, 2021-12-12:
I think there are two code issues involved here:

1) in init.ino, only the 3 cylinder case recognises that IGN_MODE_SINGLE is 'sequential' and sets for 720 degrees of crank. The others leave CRANK_ANGLE_MAX_IGN=360
2) 2 cylinder setup has 'even fire' defined as 180 degrees (which is confusing with boxer twins, which are 360 degrees apart and you'd expect that to be 'even fire')

so... 3 cylinders doesn't exhibit this 'issue' because it's explicitly coded as sequential, and 4,6,8 cylinders don't exhibit this 'issue' because the outputs are all overlaid on the single ign channel and the extra/redundant spark triggers coincide with legitimate spark triggers for other cylinders. 
I guess this isn't a practical problem even with 1,2,5 cylinder engines, because they must use a distributor to route the output sparks and the redundant sparks happen when the distributor rotor won't be connecting to an output plug lead.
Maybe this results in excess dwell on the coil for those sparks which can't be discharged?
I would think the 'proper' fix would be to code all IGN_MODE_SINGLE same as IGN_MODE_SEQUENTIAL for  all cylinder counts (and change 'even fire' for 2 cylinder to 360).

JuanNavarroS, 2021-12-12:
In answer to
> I guess this isn't a practical problem even with 1,2,5 cylinder engines, because they must use a distributor to route the output sparks

(Always talking of 4 stroke and even firing)
2 cylinder + single coil output = Yes it needs a distributor
2 cylinder + wasted spark = No it doesn't need a distributor, and should only work at 1 channel output (just 1 coil for 2 cyl). The problem is that with this setup speeduino is giving output both to channel 1 and 2, that is wrong. Additionally both channels are suffering the issues indicated in a comment by @mike501 in https://github.com/noisymime/speeduino/pull/578#pullrequestreview-660263652_

In fact 2 cylinder + single coil output should behave identically as 2 cylinder + wasted spark, the coil should fire as many times for both using just 1 ign output, not 2.

1 cylinder + single coil output = No it  doesn't need a distributor. Should generate only 1 spark per cycle = 720º (2 sparks doesnt' hurt, but is not how it should work. B&S 1 cyl with magneto in flywheel works like so)
1 cylinder + wasted spark = No sense, and in general, for 1 cyl, other that single coil output have no sense in my understanding.
5 cylinder + single coil output = Yes it always needs a distributor

In answer to
>  the redundant sparks happen when the distributor rotor won't be connecting to an output plug lead.

Not sure for 5 cyl (neither a case that I can test), but outputs are physically just 72 deg appart in the distributor cap, a spark in the middle is at only 36 deg, plus advance (tipically 35 deg max = 17 deg @distributor) a the spurious spark can be as near as 36-17=19 deg from a output so i wouln't be confident that the spark cannot jump throug it (again just a guess)
Thanks for your support

JuanNavarroS, 2021-12-13:
In response to https://github.com/noisymime/speeduino/pull/744 comments

Even firing engine is the one that have sparks (thus cycles for each cylinder) evently spaced (stagered) for all cylinders.

Even fire in 2 cyl engine, means that a complete cycle 720º / 2cyl = 360º firing, one spark at 0º, the other at +360º next at +360º = 720º=0º, cycle completed. Even spacing of 360º between sparks. Parallel twin
[image]

Not to confuse 2 cyl boxer (aka flat twin or opposed cilinder engine) where "crank angle" between cylinders = 180º, with the "180º firing" naming
As you know, any engine should be "straigthened" to understand the real firing order, and a flat twin, when "straightened" (that is, to imaginary twist 180º the 2nd cylinder and its crankpin over the first cylinder to make it straigh), results in a straight twin cylinder with the two crank pins aligned, that is a 360º firing even engine.
Engine inmediately above and below are exactly the same for firing order.
[image]

180º firing 2 cyl engine is per definition odd, as there is a spark at 0º, next at +180º, next at +180º=360º, next at +540º = 720º, cycle completed, sparks are not evently spaced (180º, 540º). Inline twin
[image]

If you put a wasted spark coil in a 180º firing 2 cyl engine you will have 2 wasted sparks, one at BDC end of admision (I would not recomend a spark with cylinder filled of mixture, even without compressing), the other one is not as worriying, as is BDC begining of exhaust, I don't know of any engine on this wasted spark configuration, but I don't know all cases.

While if you put a wasted spark coil in a 360º firing 2 cyl engine, you will have the 2 wasted sparks both at TDC end of exhaust (no flamable mixture), it's known to be commonly used, proved and tested, in fact is what happens in cylinders 1 and 4 (or 2 and 3) in a straight 4 cyl engine.

The tunes may brake for updating the firmware under many circumstances, but who's updating the firmware for an already perfectly running engine should be aware of the setoffs vs improvements, and the ones that are updating the firmware because of bugs, are expecting bugs to be solved, so all should be aware and happy of the improved result of upgrading.

Thanks for reading

noisymime, 2021-12-13:
> 180º firing 2 cyl engine is per definition odd, as there is a spark at 0º, next at +180º, next at +180º=360º, next at +540º = 720º, cycle completed, sparks are not evently spaced (180º, 540º)

This is only true when firing sequentially though. The spark outputs of a 180* wasted spark engines go 0->180->360->540, all evenly spaced. Regardless of the actual engine config, the spark outputs are even, which is what the tune is being set up for. 

Honda and Kawasaki have both made 180* degree wasted spark twins, so they are definitely out there.

JuanNavarroS, 2021-12-13:
Now I believe I understood, and probably that's the origin of the issue, this way of managing the wasted spark is generating the problem, let me explain to see if I'm not confused.

Talking of wasted spark from here on.
In a 4 cylinder engine you generate 4 sparks, with 2 coils along 720º, as should be.
But to fullfill Honda and Kawasaki requirement its required to generate 4 sparks, with 1 coil along 720º... but hey, you have only 2 cylinders... so there should be only 2 sparks

So when I go to a more conventional (even) 2 cylinder engine (boxer), with that "odd" configuration (pun intended) there are  4 sparks instead of the 2 needed, and 2 of them are at BDC.
Not sure of the next but: Additionaly, one of the sparks is inferred, or interpolated (not the right words probably), while the other is generated by the trigger event of the distributor, so you are loosing definition at the interpolated spark event.

Imagine that 2 cyl 360º engines are just half 4 straight in line cylinder engine, pick pistons 2 and 3... in wasted spark means one coil, 2 fires, not 1 coils 4 fires.

Hope I explained where's for me the mismatch. In my mind the base engine should be the even (Being also my particular configuration, may look that I'm biased, but not, i believe is more logic, manageable and flexible), and the particular others should be the odd, also because there is no way to delete the extra sparks introduced by the firmware, but probably there is a way to add more sparks (in basic distributor, adding a couple of extra "lobes")

Thanks.

jonbills, 2021-12-14:
Juan, please can you test the branch for PR 743? then Josh can decide to include or reject the PR. 
744 (the PR for change of even fire definition) is already rejected and closed. 

I've said to you a few times that none of the sparks are directly triggered by an input trigger event, but I don't think you believe me. Please examine the code for yourself - it is not the case that one spark is triggered by the distributor input and the other inferred. the logic to read the input teeth and know the speed and relative location of the crank is decoupled from the logic that decides when to fire output channels.

JuanNavarroS, 2021-12-19:
Results from testing:
Common configuration for all test: basic distributor, even fire, Trigger angle = 20*, Locked timing =ON, fixed angle = 0*
(Graph colors: yellow = trigger at dizzy, blue = ign channel 1, red = ign channel 2)

I can only comment test results for 2 cylinder, 4 strokes even engine (360*) and the results are not correct to my eye.

1 cylinder, 4 strokes, output = single channel:
[image]

1 cylinder, 2 strokes, output = single channel
[image]

2 cylinder, 4 strokes, output = single channel
Note that  the output messed up, ignitions not evenly distributed between trigger events. 
[image]

2 cylinder, 4 strokes, output = wasted spark
[image]

2 cylinder, 2 strokes output = single channel
[image]

2 cylinder, 2 strokes, output = wasted spark
[image]

5 cylinder, 4 strokes, output = single channel
[image]

jonbills, 2021-12-20:
that looks good to me, what I'd expect for  #743 -
1 cylinder single channel output 4 stroke - a spark every 720 - looks correct (agree?)
no change for 2 stroke engines.
no change for wasted spark setups.
2 cylinder 4 stroke - that looks like what I'd expect. it's 180 fire (since not include #744), but 720 cycle. so there's a spark (just before) 0, then one at 180, then nothing at 540 and then another spark at 720/0.
 5 cylinder single channel output 4 stroke - looks correct (agree?)
as above, #743 only make a difference to 1, 2 and 5 cylinder 4 stroke engines with single channel output.

JuanNavarroS, 2021-12-20:
Hi, for 2 cyl 180*, while your apporach is for me theoretically ok... I'm unaware if is there really any engine of this kind with single coil and distributor. I can not talk from practical prospective, as I don't have first hand knowledge in odd 2 cyl engine (180*, 270*) assembly

And again, I know that you know already, this type of engines are Odd, but the configuration is set as Even, so for me no sense to have to configure the 360* as odd, while the 180* as even, and what about the 270*?, is this what would be solved with #744?

For the others indicated as "looks correct" (1 and 5 cyl), afaik at least from theoretical prospective, yes, they seem ok now.

jonbills, 2021-12-20:
I think we all agree that if we were starting from scratch, even might be set at 360. But we're not and there are existing installations that are 180 and are set to even, and existing installations that are 360 and set to odd. fixing the 'bug' would break the existing installations so the fix won't happen.
There is no real downside to setting odd 360 for a new installation.

mike501, 2021-12-20:
Is the solution to update the wiki and provide an explanation and relevent
settings for each cylinder/firing combination?