speeduino/speeduino #750: Feature Request: IAT vs IACV correction table

choyr, 2021-12-17:
Hi. when IAT goes high in summers up to 70-80C, the idle goes down (idle open loop control). If we have a IAT vs IACV correction table, we can increase the IACV opening when IAT goes up

pazi88, 2021-12-17:
Have you monitored your AFR while that happens? Usually there is heat soak or just not good enough IAT calibration, that causes lean AFR which again causes lower idle RPM. You can fix this by adjusting IAT Density curve. And also you can use the idle advance feature to have much more stable idle RPM with open loop idle control mode. More advance when idle RPM is too low and less advance when idle RPM is too high. And you get easily rid of such problems.

choyr, 2021-12-17:
yep, AFR goes lean up to 17, so i changed the IAT density curve. for the ignition advance, i can only set a max ignition advance, i dont want to idle at too high ignition..
summer in middle east is around min 70C IAT due to ambient temperature and the AC condenser heat
currently im adjusting via the IAT density curve, but i thought a IAT vs IACV table is also a better way to control it

pazi88, 2021-12-17:
Set the ignition advance "too low" in normal conditions. Like for example my engine runs 15 degrees of advance normally. But I have adjusted that to 5 degrees and increased IAC duty accordingly to keep idle RPM in correct spot. Then there is 10 degrees of advance in disposal for the idle advance feature if needed to keep idle RPM stable.

But anyways, you first need to make sure that you get the AFR to stay stable with different IAT's. Once that's done, then you can start working on the other corrections.

muuris, 2021-12-17:
Ambient temp won't be as high as 70°C so it's more of a heat soak effect. Is it possible to change IAT sensor position to get more accurate actual IAT temp? As your AFR goes lean, it isn't reading correctly so better than making another correction is to make the sensor value more accurate.

Does CLT also climb a bit in these conditions? I've dealt this with increasing IACV duty on higher than normal CLT, and idle advance in addition to that. It doesn't really need much more advance to increase idle and move to another operating spot in maps.

choyr, 2021-12-17:
> Set the ignition advance "too low" in normal conditions. Like for example my engine runs 15 degrees of advance normally. But I have adjusted that to 5 degrees and increased IAC duty accordingly to [...]

I can lower the ignition, but im making the map as oem as possible.
The iat density curve theory becomes reversed if i enrich it at high iats, instead of lowering it.
The best way is to use the closed loop idle control if it PID works perfectly, but in my experience, the open loop idle control is easier with idle advance.
The iat vs iacv correction is just an improvement for the open loop idle control

choyr, 2021-12-17:
> Ambient temp won't be as high as 70°C so it's more of a heat soak effect. Is it possible to change IAT sensor position to get more accurate actual IAT temp? As your AFR goes lean, it isn't reading [...]

Ambient is 50C, but heat soak goes up to 70 to 80C when idle at traffic with AC.
Afr sensor is calibrated correctly
Clt goes up higher

shiznit304, 2022-01-08:
I've been wanting to code this, but not sure if i should turn the IAC curve into a 3d table or just have a separate  IAC corrections table

shiznit304, 2022-01-09:
[image]
I finished writing the code. I'll go test it in my car, then submit PR for you to test @choyr

choyr, 2022-01-09:
Yeees! Thats the feature Im talking about! Great. Once Im day off, i will try that. Thank you very much. Now we have to wait when summer comes.