speeduino/speeduino #760: Future request; semi-sequential fueling on cranking with full sequential setup

JaapStruyk, 2022-01-05:
alcohol e100/m100 fuels start easier with semi-sequential on cold start, probably due not fully vaporizing the large amount of fuel with one large shot.

pazi88, 2022-01-05:
This PR does that: https://github.com/noisymime/speeduino/pull/478

JaapStruyk, 2022-01-05:
Partly, VitorBoss patch is ment for cam sync loss/missing pulse and also sets ignition to semi, have made a rude rewrite but didn't test it yet.

JaapStruyk, 2022-01-29:
Stripped down @VitorBoss #478 so fuelling is semi-seq on cranking and on half sync so alcohol fuel has a bit more time to vaporize which improves starting when cold.
Only thing not correct is injector open time mS value in TS thats still showing the same as with sequential...

[Enable semi sequential fueling on half sync and cranking.zip](https://github.com/noisymime/speeduino/files/7963857/Enable.semi.sequential.fueling.on.half.sync.and.cranking.zip)

digmorepaka, 2022-04-18:
https://github.com/noisymime/speeduino/commit/4ce8c28c5a6bd166fdb81c50ca639e59fb94ae39 issue has been resolved as of this commit

noisymime, 2022-06-29:
Resolved with #478

closed 2022-06-29 by noisymime