speeduino/speeduino #764: Knock sens

Olejick404, 2022-01-08:
I would like to see a knock sensor in the speeduino project. Are there any plans to add it?

digmorepaka, 2022-01-08:
SpeedyKnock is WIP, has been for a while.... maybe a beta soon @noisymime?

shiznit304, 2022-01-10:
the knock code is already there, but hidden. it will require an external knock unit to send the signal

Olejick404, 2022-01-10:
shiznit304 :
External knock unit, on a standalone board (the signal processor eg. HIP 9011) is no problem. But SPI pins are involved (in rev. 0.4 (in my case) - is ignition output, channel 3 & 4). This will entail a redesign of the board (it is no problem too).
And problem is: knock check is relevant under certain conditions, such as: engine temperature (there is no detonation on a cold engine), RPM (different noise levels), and look for its presence at a certain angle of the crankshaft (BDTC). All this should be taken into account in the main loop  of the program.

shiznit304, 2022-01-10:
[image]
heres whats currently there

boroko, 2022-01-10:
Curious, is the issue getting a TPIC8101 or a HIP9011 to respond properly, or is it in having SPI available to run it?  Obviously Josh has already put a lot of work into it and didn't feel comfortable proceeding.

digmorepaka, 2022-01-10:
From the streams it appears that there will just be an output sending a signal about knock, all the processing done with a 328p+knock specific IC.

Olejick404, 2022-01-11:
shiznit304:
Cool... Is this a speeduino for sure? In which version of software / board ? And in more detail in which menu item are these settings (settings/tune/spark...) ? Is it possible to take a screenshot of not only this fragment? (the program window is complete)

Olejick404, 2022-01-11:
shiznit304:
Okay, how do I activate this? On which chip is it assembled? Which pins should I connect to?

shiznit304, 2022-01-11:
edit your speeduino.ini and find the line `      ;subMenu = knockSettings,          "Knock Settings"` and remove the ; 
save then update your project settings to use the updated speeduino.ini file.

Olejick404, 2022-01-11:
Ok. But these are only settings in Tuner Studio... Is there an implementation in the source code ? Reaction to the appearance of detonation (in the Speeduino. ino file)? I didn't find something...

shiznit304, 2022-01-11:
It appears to be coded already in the source code.

smb29, 2022-06-04:
TS_CommandButtonHandler
Can someone modify it, to have knock control from TS?