speeduino/speeduino #770: PWM Idle "Reverse" valve direction ignored in Closed Loop

iltHeo84, 2022-01-11:
On my 1997 MX-5 NA (later EU 1.6 model) the ISCV operates in reverse direction (more duty = less airflow).

Selecting the Reverse option in Idle Settings works if using Open Loop and/or manually testing the valve, but Closed Loop seems to ignore it, and as idle speed drops, the algorithm keeps the valve pinned at minimum duty.

Swapping Min and Max Duty % does not work (PWM stays at 0%), negative PID parameters are invalid, and setting the valve direction back to Normal makes no difference.

I resolved by issue by using the Open+Closed Loop setting, this seems to correctly factor in the Reverse option.

pazi88, 2022-01-12:
The reverse/normal setting doesn't change how the duty is affected. It changes the actual output. So in case of normal the 100% duty is output fully off and 0% duty fully off. And if you select reversed the 100% duty is output fully off and 0% duty is output fully on.

iltHeo84, 2022-01-12:
> The reverse/normal setting doesn't change how the duty is affected. It changes the actual output. So in case of normal the 100% duty is output fully off and 0% duty fully off. And if you select reversed the 100% duty is output fully off and 0% duty is output fully on.

Then the issue is more complicated than it seemed, but still present. In Open+Closed Loop, the controller successfully keeps the idle stable at around 40% PWM (reversed, so i assume around 60% DC), while in Closed Loop only, PWM is fixed at 32% (min duty) at lower idle speed, as if the PID was trying to close the valve even more.

Tjeerdie, 2022-01-14:
And to be honest closed loop only seems outdated to me. I recommend open+closed loop for all new tunes anyway. You can do exactly the same + more and much better results with Open+closedloop. 

I should have converted all closedloop  setup to open+closed loop with openloop all 0% duty cycle TPS at 100% and RPM hysteresis at maximum RPM. That has the same response as old open loop.