speeduino/speeduino #773: Possible 202202 Firmware bug?

AshtonRey, 2022-01-15:
Good morning, I’ve been having a terrible misfire during acceleration. My AFRs go from targeting what I want it to, then starts to misfire and show on the gauge 9-10 AFR sporadically.  However I am able to cruise no problem and hit my 15.1 AFR target. I double checked all my settings in case I missed anything from the previous firmware. The issue started as soon as I switched to the newest 202202 firmware. I data logged my acceleration and it isn’t a false Acceleration enrichment, and my gamma enrichment stays consistent and says it isn’t adding any additional fuel. This could be my tune being the issue, but I just want to see if anyone else has reported this odd behavior. Thank you very much for the assistance. ~1991 Mazda Miata ITBs all MAPS are load-TPS (Alpha-N).

VitorBoss, 2022-01-16:
As the TPS resolution changed to 0.5% all tables had updated the load axis to match that change, please post your tune to help us understand/reproduce the error.

AshtonRey, 2022-01-17:
I couldn't upload it in Github due to file type. So here's a link to my file.io which I uploaded my tune on. Thanks for the help. https://file.io/IH0QAvGmlW7y @VitorBoss

AshtonRey, 2022-01-17:
Here is also my data log of the issue.
https://file.io/y96eljEcmYPr @VitorBoss

AshtonRey, 2022-01-18:
Haven’t heard back yet. My temporary solution is running without Accel enrichment until I find the time again to revere to the previous firmware. @felipegaspari

AshtonRey, 2022-01-18:
I’ve also tried things like loading the new firmware with a completely new tune. But still have the issue. My data logs don’t show any TPS noise etc.

AshtonRey, 2022-01-22:
So I’ve reverted back to 202003 firmware and with the same settings the Accel false trigger does not happen at all. @VitorBoss

VitorBoss, 2022-01-22:
Just loaded your tune, the first tpsDOT value is 250 but you are using a 20% threshold, this means it will apply the first value if the tpsDOT is greater than 20%/S, to fix that you need to scale the very first values to match your threshold!

Try set the fist values to 4% and 30%/S of tspDOT.

Before it wasn't an issue due to the slow sampling and small resolution for TPS, 202202 have double the resolution and sampling rate, the tpsDOT value is scaled to match that, but it will react faster to TPS movements.

AshtonRey, 2022-01-23:
oh okay, will give it a try. If you don’t mind could you explain a little more in depth what TPS threshold and TPSdot? It’s always been a little confusing. Thank you for the assistance. @VitorBoss

AshtonRey, 2022-01-23:
Also do you think my fuel load for the Y axis is a little too close together? Or is it okay? I think it’s 0.5, 1.0 etc. I was trying to get a smooth throttle tip-in. @VitorBoss

VitorBoss, 2022-01-23:
I had never used alpha-N before, the resolution for you throttle is up to you, it usually is very small on the first % as the TPS is linear but the throttle isn't.

The threshold value is the amount ot TPSdot to be ignored, you can disable very small enrichments using that. 

TPSdot is the calculated amount the throttle will move in a second, sometimes the value will be higher as it consider the sensor will keep moving.

AshtonRey, 2022-01-23:
Oh gotcha, thank you! Would you also know why MAP based Accel enrichment isn’t working? It isn’t anything mechanical like the sensor itself etc. it does read a MAP signal on the tuner studio gauge. Again thank you for answering my questions.  @VitorBoss

noisymime, 2022-04-16:
Just having a look through things as I am seeing this also on my ITB car. One thing I've noticed is that the `taeThres` is not working correctly since the 0.5% resolution change. 

Cause is in the way `TPS_rateOfChange` is being used. in `correctionAccel()` `TPS_rateOfChange` is set with:
`TPS_rateOfChange = TPS_READ_FREQUENCY * TPS_change;`

`TPS_change` is a raw TPS value, hence is multiplied by 2. However then further down we get:
`if (TPS_rateOfChange > configPage2.taeThresh)`

`taeThresh` is NOT multipled by 2, so we're comparing 2 values that are on different scales. 
I'm not 100% sure this is the cause as I need to do some more testing, but it's an issue that needs resolving regardless.

noisymime, 2022-04-16:
I've just pushed up a change related to this. Could you please try the latest master and see if you are still having the same issue?

noisymime, 2022-05-15:
Closing as this appears to be resolved with the above fix

closed 2022-05-15 by noisymime

RempageR1, 2022-05-16:
I fail to understand how this makes any change. 

The TPS_change has gone from 8-bit value (max 255) to 16-bit value (65K) but it only holds the TPS delta. Since TPS ranges from 0 to 200, it would still fit the 8-bit.

As for the other code changes, let`s do the calculations:

TPS_rateOfChange = (TPS_READ_FREQUENCY * TPS_change) / 2; //This is the % per second that the TPS has moved, adjusted for the 0.5% resolution of the TPS
30 * 5 / 2 = 75
if(TPS_rateOfChange >= 0) { currentStatus.tpsDOT = TPS_rateOfChange / 10; } //The TAE bins are divided by 10 in order to allow them to be stored in a byte
75/10 = 7.5
	
TPS_rateOfChange = TPS_READ_FREQUENCY * TPS_change; //This is the % per second that the TPS has moved
30 * 5 = 150
if(TPS_rateOfChange >= 0) { currentStatus.tpsDOT = TPS_rateOfChange / 20; } //The TAE bins are divided by 10 in order to allow them to be stored in a byte and then by 2 due to TPS being 0.5% resolution (0-200)
150/20 = 7.5

So both seem to do the same or am I missing something here?

pazi88, 2022-05-16:
Yes all good. You need % per second and that calculation gives that. The point being that you catch smaller movements (0.5% of TPS) and more frequently to react TPS movements faster than earlier.

VitorBoss, 2022-05-16:
`TPSdot` is right, the result is the same, but `TPS_rateOfChange` is used in other places and I missed that

noisymime, 2022-05-16:
The difference is in the scaling of `TPS_rateOfChange`. It’s value had been doubled due to the 0.5% resolution, but then it was compared in an if statement: `if(TPS_rateOfChange > configPage2.taeThresh)`

`taeThresh` Only has a 1% resolution, so the threshold value was effectively being halved and therefore caused many false AE triggers to occur.

RempageR1, 2022-05-16:
Perfect, thank you for the clarification!