speeduino/speeduino #774: Protocol error on TunerStudio while changing large areas of maps

sokerer, 2022-01-17:
As shown in the following pictures, on the latest firmware releases, if we attempt to change large ares of the maps with < > cursors on the keyboard, a protocol error appears, and a reset of the bard is required in order to reinitiate the com.
This was not happening with the previous firmwares, attempt was done on 3 different genuine arduino mega boards, and 2 different usb cables.
Tests were done after a fresh install with speedyloader of the latest 202202 firmware, with relative ini file, and base tune
[image]
[image]

adbancroft, 2022-01-18:
Can you try deactivating the NEW_COMMS feature in Project Properties?

sokerer, 2022-01-18:
Yes you are right, the problem disappears with the NEW COMMS deactivated !!

NEW Comms have been there since march 2021 release ?
I ask because it's from the March 2021 release, that I started to find this
problem

Corey-Harding, 2022-01-18:
I saw noiseymime recently made a new commit on master branch that may fix this if you two didnt notice

https://github.com/noisymime/speeduino/commit/fe33785ec5abaccf675df2a2debd16a793163bed

sokerer, 2022-01-18:
Yes, I just had a small test with the latest master, and looks like after Noisymine patch, the problem with the NEW COMMS is solved !!

adbancroft, 2022-01-18:
If solved, could you close this?