speeduino/speeduino #786: 202202 cannot set Temp Sensor or AFR

ColHQ, 2022-02-04:
When trying to go via Tools menu and either Calibrate Temperature Sensor or the Calibrate AFR Sensors  a "Error Showing Dialog: Null" box is displayed and cannot set or calibrate these sensors.

pazi88, 2022-02-04:
Bug in 3.1.07 TS. Use 3.1.06 instead

ColHQ, 2022-02-04:
Any idea where I can source a copy of 3.1.06 , There is nothing I can find.

pazi88, 2022-02-04:
For others looking: https://www.efianalytics.com/TunerStudio/download/oldReleases/TunerStudioMS_Setup_v3.1.06.exe

noisymime, 2022-02-04:
TunerStudio 3.1.07.1 should be incoming shortly to resolve this.

closed 2022-02-07 by noisymime

noisymime, 2022-02-07:
I've closed this record for now as c6a37f7347ad0522778d8d44c4379a0dc913dec2 resolves it in master. Previous releases however will need to use TunerStudio 3.1.06 for the moment until 3.1.07.1 is out. 

Background here is that there is a command `tableCrcCommand` that is used by TS to determine what calibrations (IAT, CLT and O2) are currently on the board. This is an optional command, however v3.1.07 unintentionally assumed it was implemented on all boards and gave a 'null' error if the command was not present. I have now implemented that command as is it nice to have anyway, but there will also be a fixed version of TunerStudio coming out shortly.

commit c6a37f73 (2022-02-07): Implement tableCrcCommand command. Fixes #786 until TS 3.1.07.1 is released

The tableCrcCommand is used by TS to determine what calibrations (IAT, CLT and O2) are currently on the board. This is an optional command, however v3.1.07 unintentionally assumed it was implemented on all boards and gave a 'null' error if the command was not present.
This adds the command for all future versions, but previous releases will need to wait for theTS fix release