speeduino/speeduino #794: Feature request: Add pins A16-A22 to Local Aux tab

jankovalski, 2022-02-20:
Right now you can choose pins up to A15 as aux inputs, even if Teensy is selected as the MCU. The only place where pins above A15 are available now are for fuel/oil pressure. Please add them to the aux input tab too.

[image]

pazi88, 2022-03-14:
If someone wants to try, this should be simple as replacing line 258 in speeduino.ini: `#define ANALOG_PIN = "A0", "A1", "A2", "A3", "A4", "A5", "A6", "A7", "A8", "A9", "A10", "A11", "A12", "A13", "A14", "A15", $PIN_16inv, $PIN_16inv, $PIN_16inv` 

With: `#define ANALOG_PIN = "A0", "A1", "A2", "A3", "A4", "A5", "A6", "A7", "A8", "A9", "A10", "A11", "A12", "A13", "A14", "A15", "A16", "A17", "A18", "A19", "A20", "A21", "A22", "INVALID", "INVALID", "INVALID", "INVALID", "INVALID", "INVALID", "INVALID", "INVALID", "INVALID", $PIN_16inv, $PIN_16inv`

noisymime, 2022-07-10:
Note that currently TuenrStudio doesn't allow you to add `#defines` from within a `#if` conditional, so this can't easily be changed based on the project setting. Will have to try and find a workaround for this limitation.