speeduino/speeduino #796: Dfco, Flood Clear and Fuel Pump Bug

omardlr, 2022-02-20:
FW Version: 202202
Board: UA4C

Hello! After updating my ecu to the latest firmware (202202), I am having problems with my car, DFCO is not working, Flood Clear is not working either, Flood Clear is activated in Tuner Studio but the ignition outputs are still working, my fuel pump shuts off if I stop the car in idle more or less for about 3 minutes, all this happened after updating to the latest version of speeduino.

pazi88, 2022-02-21:
Usually those kind of problems are solved by doing eeprom clear. And reloading the tune. eeprom clear example is available in Arduino IDE

noisymime, 2022-02-21:
I'm wondering whether this one might be related to the TPS resolution changes. I'll have a chance to test and fix tomorrow if it is.

noisymime, 2022-02-21:
Have tested DFCO and flood clear on current master and both appear to be working fine. Moving above and below the configured TPS levels works ok and causes the expected output changes. 

It's possible something hasn't updated properly during the upgrade process though. Can you try what @pazi88 suggested, clearing the EEPROM (You can do this through Speedyloader now, scroll to the bottom of the firmware list for the EEPROM clear) and then load the 202202 firmware and your tune back on?

closed 2022-07-15 by omardlr