speeduino/speeduino #799: 10Hz AFR

jackscr12, 2022-02-23:
I've seen talk about this mod countless times on slack,fb etc.... almost everyone agrees that 4Hz AFR is really slow .I've been implementing 10Hz AFR for almost a year now(on mega) and it works fine as have others so what's the hold up?

theKraid, 2022-02-23:
I was wondering about this as well! The current default refresh rate is way too low
If you have a PR I would love to test it

pazi88, 2022-02-23:
> I was wondering about this as well! The current default refresh rate is way too low If you have a PR I would love to test it

You can just move the read O2 to from 4Hz flag to under 10/15 or 30 flag.

transi01, 2022-02-23:
in Which file in the source can it be changed?

Thanx
// Thomas

pazi88, 2022-02-23:
https://github.com/noisymime/speeduino/blob/master/speeduino/speeduino.ino#L329

Move that to under different timer flag if's

jackscr12, 2022-02-23:
IN speeduino.ino you move       readO2();    from the 4HZ flag up to the 10Hz

jackscr12, 2022-02-23:
going to try 30Hz as well and see how it impacts the loops just out out of curiosity.

digmorepaka, 2022-02-23:
I have tested 30Hz on the bench and didn't see any major dip in loops/sec. Note  this  was without any closed loop functions enabled.

jackscr12, 2022-02-23:
> I have tested 30Hz on the bench with a mega and didn't see any major dip in loops/sec. Note this was without any closed loop functions enabled. IMO on the Teensy and STM it should be 30Hz by default.

Hence my question ,why is this not default at least  10HZ... PID and autotune DO NOT work ok with it at 4Hz unless at reasonable steady state or setting the PID to react really slowly. And also data-logging would be way more useful at this higher read rate

transi01, 2022-02-23:
Found it, thanx! Will test later the day... //Thomas Von meinem Huawei-Telefon gesendet-------- Ursprüngliche Nachricht --------Von: jackscr12 ***@***.***>Datum: Mi., 23. Feb. 2022, 09:31An: noisymime/speeduino ***@***.***>Cc: transi01 ***@***.***>, Comment ***@***.***>Betreff: Re: [noisymime/speeduino] 10Hz AFR (Issue #799)
IN speeduino.ino you move       readO2();    from the 4HZ flag up to the 10Hz

noisymime, 2022-02-24:
I'm in full support of moving to (at least) 10Hz for these readings, my only concern would be any potential impact on existing closed loop tunes. I expected the impact would be minor, but any decently tuned PID tune would have taken the current read speed into account. 

I'll have a play and if no major impact to existing PID loops I'll up this, probably to 15Hz

Corey-Harding, 2022-02-24:
10Hz has been a major improvement for me.  Switching from 4Hz to 10Hz made my readings less sporadic and my car run smoother, veal autotune even made more sensible adjustments.  Also my loops per second may have even increased on mega.  They were around 900 at idle and now I see around 1000.

jackscr12, 2022-02-24:
> I'm in full support of moving to (at least) 10Hz for these readings, my only concern would be any potential impact on existing closed loop tunes. I expected the impact would be minor, but any [...]

Since no one is running EGO at WOT(at least no one should as at 4Hz it's pretty useless) there is no risk of damage from changing the AFR to 15-30Hz.The minimal inconvenience of retuning PID is small in comparison to the advantages of high refresh rate AFR, I'm sure everyone would be more than  happy with the change.

pazi88, 2022-02-24:
Yeah you can't even have very well working closed loop EGO control with just 4Hz rate. At least I haven't been able and what I have heard, others too haven't had much luck with that. You can of course get it do something, but nothing compared to closed loop EGO control in other aftermarket ecus. If it's not that much of performance drop, I would go 30Hz directly and we can fine tune the PID algorithm for that if needed.

Corey-Harding, 2022-02-24:
Would it also be beneficial to try moving the idle function outside of 4Hz to 10Hz?  Or is that one part of code that would create a substantial decrease in loops?  My idle is still not perfect under all conditions running open+closed loop pid idle.

pazi88, 2022-02-24:
> Would it also be beneficial to try moving the idle function outside of 4Hz to 10Hz? Or is that one part of code that would create a substantial decrease in loops? My idle is close but still not perfect under all conditions(instances of oscillation under sudden multiple heavy acc loads like headlights+blowers full on+cooling fan+power windows) running open+closed loop pid idle.

In my experience making icv control faster and/or more agressive just causes more oscillations rather than preventing those. There is always some delay when you try to control idle rpm with air amount, so the changes/corrections in that need to be rather slow to avoid oscillation. For fast responses to correct idle rpm, idle advance is way much better way. You only need closed loop for correcting steady state error. Usually there is no need for P-term in closed loop idle control, only I-term is needed and slow rate for that is best.

Corey-Harding, 2022-02-24:
The best settings I was able to achieve was at almost 0 P/D and a lot of I.  So what your saying sounds spot on, I may need to work on my PID more or tweak my idle advance more. Its really close to oem except under extreme acc loads it dips and then overshoots a bit before oscillating back into place. Under normal conditions it is nearly an oem experience. Once I get a chance to try reado2 at 30Hz I may tackle pid ego as well.  So far at 10Hz its feels more like I just upgraded the hardware rather than a simple firmware update.

Corey-Harding, 2022-02-24:
I just noticed PR #784 is running readO2 at 30Hz and like you said idlecontrol stays at 4Hz

MykkH, 2022-02-24:
Is it possible to do the same for MAP reading?

pazi88, 2022-02-24:
> Is it possible to do the same for MAP reading?

MAP is already sampled once every loop (900-1000 Hz)

MykkH, 2022-02-24:
> Must be the way the log is grabbing the data then, in MLVHD the MAP updates at twice the speed of the O2, so I presumed somewhere around 8hz

pazi88, 2022-02-24:
If you are on unregistered TS, max data rate you will get out of speeduino is 15Hz. Also in case of a map signal, the used filtering plays a role of how often the reading is updated. In cycle average mode, the reading is updated once for every cycle (all the samples are averaged during cycle period). In instantaneous mode, filtering amount will greatly affect how often map values seems to update .

npmstart-pray, 2022-02-24:
The higher the sampling rate, the more accurate the waveform data. CD audio is 44100Hz (for comparison) and even that makes for differences between reproduced waveform and original. We should be sampling at 1MHz or higher on all sensors. Then you can really get a read of what the 1v or 5v sensor signal waveform is saying, and that leads to better tuning for power/efficiency. Air-fuel metering, spark timing, shift points...everything gets more accurate with a faster/higher sample rate.

pazi88, 2022-02-24:
Well faster is usually better,  but not all sensors need to update super fast. In example iat and clt readings especially don't change that often. There is no really benefit on going to megaherz or even kiloherz range on the analog readings in a ecu. Also in case of mega especially, no processing power to handle all the data in that rate. There is no really need to capture waveforms in ecu. We only need to know enough data to calculate fuel and ignition.

npmstart-pray, 2022-02-24:
Fuel needs air, exhaust is the check it was right, and coolant temp can cause overfuelling to cool the engine. Ignition timing affects power/efficiency. It’s a Wizard of Oz scenario, the ECU being the man behind the curtain nobody pays attention to, but a lot is happening and needs to be minded appropriately. MAP, MAF…everything. If current hardware is insufficient for higher speeds, what is there to upgrade to? RasPi nano/zero as sub processors feeding speeduino? If so, what clock do they all reference? (What’s the throughput capacity of CanBus? Or are we looking at building proprietary, ground breaking OBD3?)

Sent from nowhere near my ready room with my new tricorder

VitorBoss, 2022-02-24:
> I'm in full support of moving to (at least) 10Hz for these readings, my only concern would be any potential impact on existing closed loop tunes. I expected the impact would be minor, but any [...]

Current idle PID runs at 4Hz rate, the sensor reading speed won't impact the control, it will just produce a faster update on TS gauge and improve TS auto tuning.

I have use it for months in 100Hz update rate without any impact in idle control

jackscr12, 2022-02-24:
I don't know what you are talking about.. 1Mhz refresh rate for sensors,...afr sensors do not even react that fast never mind their controllers..100HZ for AFR would be ideal but 30hz is good enough...  CLT and IAT... if you look at the corrections affected by those in the code you will see that having a high refresh rate will amount to seeing oscillation of a tenth of a degree in temp and that is going to affect the actual fuel being injected by next to nothing...so what is the point 1Mhz refresh rate? or even 100HZ..when it takes CLT,IAT a full second to change a tenth of a degree.

noisymime, 2022-02-24:
> We should be sampling at 1MHz or higher on all sensors. Then you can really get a read of what the 1v or 5v sensor signal waveform is saying

Let's get some perspective as there's some serious armchair engineering going on here. To claim a 1mhz afr read rate would be useful is just crazy and shows a complete lack of understanding for how any of this works. 

This chart has been around for a few years and whilst it is a marketing piece and should be treated as such, the data in it is accurate
[image]

Since the testing was done, there have been faster wideband controllers hit the market, bit it's safe to say that anything using any of the typical reference designs will be seeing average response times of around 80-100ms. 

Based on this data I seem to think somewhere around the 15hz mark is going to work best.

VitorBoss, 2022-02-24:
Using 100Hz I had log updates as fast as 20ms on my APSX D1 controller. The differences where very tiny, could even be noise tho

closed 2022-02-25 by noisymime

RempageR1, 2022-05-30:
Does this change also impact the Analog sensor filter value for O2? The current recommended value is 128.

pazi88, 2022-05-31:
> Does this change also impact the Analog sensor filter value for O2? The current recommended value is 128.

Yes it does affect. Though I personally have the filter at zero in the setups I have done, because the O2 sensor signal isn't really noisy. It gives the fastest responding O2 value.

commit 87e21fc7 (2022-02-25): Increase O2 read frequency to 30Hz

Fixes #799