speeduino/speeduino #809: EGO selection possibly not storing correctly

ColHQ, 2022-03-06:
I am having a problem where I select an EGO in the Tools/ Calibrate AFR option (14point Seven), write it to the controller. Then close the window and reopen to verify selection, the field that shows the EGO type is blank. I have tried going back to Tunerstudio v3.1.06 to no avail. Have also tried changing Speedy firmware revisions, no good either. So not sure if the correct setting is being written.

Note:  I am not connected to the shield board or in the car as away at work. Just using an Arduino board connected to my laptop to do my tune checking, in case that makes a difference. Though I know it used to display like this, as I have done a JPG snip of all the settings windows as a reference.

Cheers
Col

digmorepaka, 2022-03-06:
Does the analog  input behave like it should?

gdiciocco, 2022-03-06:
I'm using it with 2 probes and it is behaving correctly. Although not showing any value or choosen choice.

noisymime, 2022-03-07:
The selection of the sensor isn't stored in the Tune itself. When you select one of the sensors, TunerStudio is generating a table of values and sending those to the ECU, but it doesn't actually store which selection you made, so the behaviour you're seeing is normal. 

The current Dev firmware has implemented a checksum so that TS can try and reconcile the values in the ECU with its list of sensors. It's a little clunky, but helps address the concern you're having. Worth noting though that TS only supports this feature with the new comms protocol. 

Closing this for now as the described behaviour is normal for the current release

closed 2022-03-07 by noisymime