speeduino/speeduino #810: Feature Request: Ford TFI Trigger option for sequential

hgomez09, 2022-03-08:
Good evening, 

Myself and a few other dropbear owners are looking to add an option for TFI ignition in the speedy firmware. I'm not an electrical engineer, but from what I understand, the TFI distributors use an internal trigger wheel with varying gaps and a "PIP" sensor to properly locate cylinder 1 then uses the TFI module mounted to the body to send the signal to ECU and accept the spark signal to fire the coil. I guess this is similar to Dual Sync..? Check this link below.

http://www.ozfalcon.com.au/index.php?%2Ftopic%2F8919-the-ford-tfi-ignition-system-bible-tech-article%2F&fbclid=IwAR1n0xX4jZ_FhIR69lxXZqc2scjdNRFqAOmoOE54eLY2PLQArHUcNa9qPN0

It does allow for sequential fueling in the 5.0 mustangs, thunderbirds, mark vii's f150's, etc., and we'd like to have this added so we can use the dropbears on our projects to it's full potential. I'll try to compensate anybody who can help! Thank you!

DeionSi, 2022-03-08:
I believe what you are referring to is called "Signature PIP" for TFI. Some info at https://www.facebook.com/DragRadialPerformance/posts/heres-a-picoscope-waveform-of-what-ford-tfi-ignition-signals-look-like-the-blue-/1919035208189680/.
In terms of decoding it is essentially a "short tooth" instead of a "missing tooth".

hgomez09, 2022-03-08:
Exactly right, and this is what MS3 ecu's call it as well. :)

hgomez09, 2022-03-17:
More info provided by Mike501 on discord. 

https://forum.arduino.cc/t/ford-signature-pip-signals/496816

Per Mike: "The code shouldn't be too difficult. If you compare the falling edge previous gap to the current gap. Tooth 1 should be 80 percent of that value. Using bitwise maths you could do the check on greater than 12.5 percent so anything less than 12.5 percent is engine acceleration/ deceleration. Gaps greater than 12.5 percent is tooth 1. More than 8 teeth and you've lost sync."