speeduino/speeduino #815: digital ports don't work as expected using Teensy 3.5 (actual daily FW)

transi01, 2022-03-14:
As for example for D30 and D34 on teensy 3.5 the output of the pins when switching to high is about 0,75V so these ports are not usable as output channels.

I'm using my own designed teensy 3.5 adapter board but with firmware 202108 at least D34 was working.

Thanx for help!

// Thomas

Corey-Harding, 2022-03-14:
Try changing the value on line 100 in sensors.ino from 63 to 127 and let me know if it works. I don't have any digital outputs but changed the value to 63)+1 recently in a PR because it matches a line in speeduino.ino I believe it was. But maybe the lines in sensors.ino for pina and pinb were set to 127 for a reason in sensors.ino.  I'm having trouble as well with an analog progout I have for AC control activating for a few seconds then turning on and off when the switch is on. I haven't had time to break out a multimeter though and get in depth with whats causing it but I also have an issue open.  It was working a few months ago though or at least I thought it was at the time but it definitely isn't now.

transi01, 2022-03-14:
should I still try to change value in line 100 of sensors.ino from &63 to &127?

Corey-Harding, 2022-03-14:
I don't think so.  My CEL output works fine.  Its my A8 INPUT that doesn't work right.

pazi88, 2022-03-14:
&63 and &127 difference doesn't really affect those. 

From tune I can see the board is "Productions PNP Boards". How does your board config look like in init.ino? Assuming this is custom board.

Also the nitrous arming pin is 34, so it might set that pin as input in FW

transi01, 2022-03-14:
[Board_Layout.txt](https://github.com/noisymime/speeduino/files/8246595/Board_Layout.txt)

This is my board layout. Nitrous is disabled, so it should not affect D30 and or D34 I guess?!

But in between I have changed nitrous pins to 27 without any change to the behavior while switching D34

It goes from 0V to 0,75V but not to the 3,3V which it should set the high level to using Teensy3.5

/ Hope that makes it a bit clearer

transi01, 2022-03-14:
[speeduino_Power_CPU.txt](https://github.com/noisymime/speeduino/files/8246659/speeduino_Power_CPU.txt)

This is my speeduino.ini file for the custom board

transi01, 2022-03-14:
the strange thing for me is that I can access the D34 pin and the indicator goes to active so it should not be blocked in FW as an input or so I guess?!

But when I't beeing active verified using TS indicator for "Programmable out 4" it goes from 0 Volt only to 0,75V...

pazi88, 2022-03-14:
The high state not reaching full 3.3v (or 5v) is usually because same pin is defined as input somewhere else.

transi01, 2022-03-14:
hmm... strange, but now Pin 34 is no longer defined elswhere than for "Programmable output 4"

and aren't are the as IGN defined pin's locked... cause same behavior for D30 which is defined as IGN3

noisymime, 2022-03-14:
> But when I's going active (verified using TS indicator for "Programmable out 4") it goes from 0 Volt only to 0,75V... 

This is very typical behaviour for a pin that has been set as an input rather than an output. The firmware is setting its value to high, but it's mode has not been set correctly. 

Just as a quick look through your tune, you've got the `n2o_arming_pin` (Which is an input) set to 34, although it's turned off so it _shouldn't_ be impacting things. It would definitely be worth changing this though to see if it makes any difference.

transi01, 2022-03-14:
In the meantime I have already changed n2o_arming_pin to another output saved config and disabled nitrous again... Behavior is still the same :-(// Thomas Von meinem Huawei-Telefon gesendet-------- Ursprüngliche Nachricht --------Von: Josh Stewart ***@***.***>Datum: Mo., 14. März 2022, 22:14An: noisymime/speeduino ***@***.***>Cc: transi01 ***@***.***>, Author ***@***.***>Betreff: Re: [noisymime/speeduino] digital ports don't work as expected using Teensy 3.5 (actual daily FW) (Issue #815)

But when I's going active (verified using TS indicator for "Programmable out 4") it goes from 0 Volt only to 0,75V...

This is very typical behaviour for a pin that has been set as an input rather than an output. The firmware is setting its value to high, but it's mode has not been set correctly.
Just as a quick look through your tune, you've got the n2o_arming_pin (Which is an input) set to 34, although it's turned off so it shouldn't be impacting things. It would definitely be worth changing this though to see if it makes any difference.

transi01, 2022-03-15:
... have just created a complete new project using a new teensy 3.5. Same result for Port D30 and D34.

Is there any way to set a port to output mode while compiling firmware?

closed 2022-03-21 by transi01