speeduino/speeduino #818: Civic 92 MAP sensor can't be used because of high values at 0 volts

Matakratzka, 2022-04-02:
It is not possible to use the MAP sensor from the 1992 Civic because it has too high pressure values at 0 volts. 
This has a value of 203 kPa at 0 V and 15 kPa at 5 V, but the maximum value at 0 V is 127 kPa in tunerstudio.

I've attached a picture of the pressure diagramm. The linear function is f(x)=-37.627kPa*x+203.473kPa.

[image]

noisymime, 2022-04-02:
Just to confirm, have you verified on the sensor itself that it outputs these readings? I've never seen an automotive MAP sensor that runs from high to low before is all.

Matakratzka, 2022-04-03:
I've measured the voltage today and you're right!
There is 0.8V at idle and 2.9V at atmospheric pressure.
Thanks for your help.

closed 2022-04-03 by Matakratzka