speeduino/speeduino #819: Use 200MHz RTL8722DM?

poky, 2022-04-02:
Hi, I'm wondering what's the requirement spec for the Speeduino mcu?

I came cross to this Realtek RTL8722DM board, spec as following: 
https://www.amebaiot.com/en/amebad/#rtk_amb21

ARMv8M Real-M300 Cortex-M33 compatible MCU @ 200MHz
ARMv8M Real-M200 Cortex-M23 compatible MCU @ 20MHz
Max. 55 GPIOs
4 Hardware timers
Also Wifi + BT5.0

[image]

noisymime, 2022-04-02:
Really there's 2 things to consider

1. I'd need to have a read through the datasheet (Which annoyingly appears to require a sign-up) to see what the configuration of those timers is (Mainly how many channels each one has). 
2. Support for the architecture within PlatformIO. This isn't strictly needed, but it will be an enormous pain to maintain without it. There appears to already be a request for this, but it hasn't gone far: https://github.com/platformio/platformio-core/issues/769

poky, 2022-04-03:
Thank you for reply!

As for the datasheet, I hate they put the login requirements on a public open docs.
But have found this from the seeedstudio.
https://files.seeedstudio.com/products/102110419/Basic%20documents/UM0401_RTL872xD_Datasheet_v3.4_watermark.pdf

Timer spec as following:

[image]

Hope these helps!

noisymime, 2022-04-28:
That should be plenty of timer channels, so no issues from a hardware front. 

The main hurdle now would be getting support for this MCU in PlatformIO. There doesn't seem to have been much action on this on the above linked issue, but if that can be achieved then support in Speeduino should be fairly straighforward

noisymime, 2024-03-15:
Closing this for now as there is currently no path to implementing this until there is support within PlatformIO.

closed 2024-03-15 by noisymime