speeduino/speeduino #821: Serial Monitor in Arduino IDE not functioning properly with 2022.02 firmware

debrucer, 2022-04-02:
Works with previous release (2021.08) but not functioning in the latest. Enter the ? to activate the menu and no response. Repeated entry gets garbage like it's the BAUD rate, but that is set properly. Enter the command NSNN one or more times and you sometimes get the version number. No other commands appear to work. Both versions seem to work properly in Tuner Studio. This is happening with Teensy 3.5 and Dropbear with same MPU. Repeatable on more than one device. Have not tried with Mega 2560

debrucer, 2022-04-02:
Tested Mega 2560 with same result. Works with prior version. Does not work with current version.

Here's picture of the menu that I am referring to:

[image]

dantob, 2022-04-03:
I bisected this once before in discord, the culprit commit was the initial new comms (although I do notice now, that it says disabled by default). Try a git bisect and see if you get the same results.

```db2e667abb41da714aa1c09bafbf9a5ddf7ab46b is the first bad commit
commit db2e667abb41da714aa1c09bafbf9a5ddf7ab46b
Author: Josh Stewart <josh@noisymime.org>
Date:   Fri Nov 19 11:07:38 2021 +1100

Initial work on new serial comms (Disabled in ini)
```

debrucer, 2022-04-03:
I'm not in the position to try anything this evening, I brought the subject
up on Slack (help thread) and had a conversation with a regular there about
"flags"., which aren't required for teensy compiles, but they were set.
I've also tried all of the performance options on the compile, all with the
same result. (a couple of warning messages in there, too, but warnings).
Guess I've got to get on Discord for the searchability not available on
Slack. I'm not hard-pressed for a fix, but did want to report the problem.

VitorBoss, 2022-04-03:
What is this about? Only the response to `?` Request?

Try to send `Q` it will answer correctly.

debrucer, 2022-04-03:
@dantob I'm not in the position to try anything this evening, I brought the subject
up on Slack (help thread) and had a conversation with a regular there about
"flags"., which aren't required for teensy compiles, but they were set.
I've also tried all of the performance options on the compile, all with the
same result. (a couple of warning messages in there, too, but warnings).
Guess I've got to get on Discord for the searchability not available on
Slack. I'm not hard-pressed for a fix, but did want to report the problem.

noisymime, 2022-04-03:
This is to be expected with the `?` command as it was never considered in the move to the new serial comms. It can be added back in fairly trivially though.

As mentioned, other legacy letter commands should work though.

commit 0dc31edc (2022-04-11): Allow for '?' command through legacy comms.

Fixes #821
closed 2022-04-11 by noisymime