speeduino/speeduino #822: Feature request: Low battery and alternator indicators

Corey-Harding, 2022-04-08:
For those with dashes running tuner studio a user configurable low battery indicator may be useful when engine is off may be useful as well as an alternator low/high warning when engine is running.

Fera31, 2022-04-08:
you can use a programmable output to do that

Corey-Harding, 2022-04-08:
> you can use a programmable output to do that

I am referring to lighting up a battery and alternator icon on a tuner studio digital dashboard and not a physical light on a traditional analog dash.

brunob45, 2022-04-09:
For something simple, you could add an indicator directly in tunerstudio. Something like that would work (in speeduino.ini,  around line 1450):
 
 ​   ​indicator​ = { batteryVoltage<11     }, ​"​Battery low"​,   ​"​battery ok"​,       red, black, white,    black

Corey-Harding, 2022-04-09:
Something like that is what I was looking for.  Thanks

> Message ID: ***@***.***>
>

Corey-Harding, 2022-04-09:
Going to try:
`indicator = { (batteryVoltage < 11.8) || (batteryVoltage > 15) }, "Battery Warning", "Battery OK", red, black, white, black`

commit f8e74224 (2022-04-09): Add low/high battery voltage indicator (#824)

Fixes #822
closed 2022-04-09 by noisymime