speeduino/speeduino #829: Warning in TunerStudio after [WIP] CAN broadcasting menu (#639)

Corey-Harding, 2022-04-21:
```

1 Warnings:
Warning: Failed to set value for unusedClusterBits .msq value is not valid for current configuration: No valid options found for Bit EcuParameter:unusedClusterBits equal to the proposed 0

```

Corey-Harding, 2022-04-21:
From my msq tune file
```

<constant name="canBMWCluster">"On"</constant>
<constant name="canVAGCluster">"On"</constant>
<constant name="enable1Cluster">"On"</constant>
<constant name="enable2Cluster">"On"</constant>
<constant name="unusedClusterBits">0</constant>
```

I flashed firmware before this commit
[Remove duplicate updates from last PR](https://github.com/noisymime/speeduino/commit/97f8ef795a514c9a5693e85be6cdb8bdee8ef2c5)

So I assume I need to turn the first 4 values to off?

floods57, 2022-05-15:
Hi, 
Thanks a lot, that worked for me ;)

commit 9673dd8f (2022-05-16): Add valid option for unusedClusterBits in ini. Fixes #829
closed 2022-05-16 by noisymime