speeduino/speeduino #833: Injector misfire - appears related to secondary trigger.

Speshman, 2022-04-26:
Copy of tune and a recent log - https://spesh.net/TuneLog26042022.zip
The tune is nowhere near finished but, aside from the issue outline below, runs ok.

0.4.3c board

Peugeot XU10J4RS engine.  60-2 primary trigger via VR conditioner.  Poll level secondary via Hall.  Wasted COP (Vag coils, 15ohm resistors) and sequential injectors. Alpha-N with jenvey ITB.  Sync gained straight away (I do have a little noise casuing a rare loss but its 99% there, not got as far as finding the noise yet)

Issue - I'm seeing a missfire/failure to trigger on injector 3 (Cylinder 4).  This presents as a "running on 3" feeling while trying to drive and is also visible on the onboard LED.  At 2000rpm approx the other injector LED's are effectively pinned on as they are flashing too fast to discern however you can see injector 3 led flash suggesting its out of sync with the others or dropping pulses.  If I invert the secondary trigger to rising instead of falling the issue moves to injector 2 (Cylinder 3).  Also, as suggested on a facebook post linked below, i've tried changing the close timing and this causes the issue to move to another injector.

Pictures relating to secondary trigger  -  I've tried running it as single tooth but no sync, i came to the poll level through trial and error and assumed its good given that I get full sync.
https://speeduino.com/forum/viewtopic.php?p=54459#p54459

Initially I thought I had a bad joint on the collector on the fet for injector 3 so i resolderded it, no joy.  Replaced the fet just in case, problem persists.

Manually triggering injectors via hardware testing is always without issue and succesful.

Someone who had a very similar issue that was resolved by an adjustment to secondary triggers (not available to me)  - https://www.facebook.com/groups/191918764521976/?multi_permalinks=1660414874339017

more info on the car/setup etc  https://speeduino.com/forum/viewtopic.php?f=19&t=5062

pazi88, 2022-04-26:
The poll level decoder is bit unfinished on my part, where it gets sync and no sync losses even without cam signal at all. If you don't get sync with singme tooth cam trigger, it means that you don't have calid cam signal available and that's your reason for injector missfire.

nuosu, 2022-04-26:
I had a similar problem with random engine misfires. But I haven't figured out which cylinder it is. This problem started with the new firmware of 2022 and I had suspected it was a problem with the board. But after I tested different boards, still the same situation. So I tried the old firmware. The engine misfire disappeared. I suspected it was the crankshaft 60-2 trigger, so I switched to the 48-2 cam signal. The problem still exists. Maybe you can try the old firmware and see if the problem is solved.

gdiciocco, 2022-04-30:
I'm on 36/2 trigger with cam reference on a vtwin. Im successfully using poll level cam from a couple of months for quicker startup. But i were probably experiencing some random misfires at low gas opening, a brief stumble, in situations where accel enrichment wasn't involved. Fallen back to single tooth cam trigger yesterday after reading this issue and the problem seems to be gone or heavily attenuated. 

To me it looks like (first rough impression) that the problem is heavily amplified by the poll level cam, but not generated by it)

pazi88, 2022-04-30:
> I'm on 36/2 trigger with cam reference on a vtwin. Im successfully using poll level cam from a couple of months for quicker startup. But i were probably experiencing some random misfires at low gas [...]

How does the cam trigger wheel look like? The poll level decoder only works well with the "half moon" cam trigger wheel.

Anyways, now that the semi-sequential/wasted spark fallback is included in FW, I'm working on getting that to work with poll level cam decoder too. In case of faulty cam signal, you would then fall back to half sync and not get just random injector misses.

Speshman, 2022-04-30:
After solving the issue (that I wasn't sure I had) with my hall sensor on my cam and setting it back to single tooth cam I can confirm full /proper/ sync and good running. 

Not related but the issue I had with the hall sensor was the VR conditioner was pulling the signal down to 2.6v even though it wasn't in use.  Removed the pins and its back up to 5v and working well (I need the conditioner as crank is VR)

closed 2022-04-30 by Speshman

gdiciocco, 2022-05-01:
> How does the cam trigger wheel look like? The poll level decoder only works well with the "half moon" cam trigger wheel.
>
>
>
> Anyways, now that the semi-sequential/wasted spark fallback is included in [...]

Need do to some logging and see if theres something happening, from last time i did there were no signs of triggering problems.

My cam trigger wheel is not halfmoon like, its just a little metal cube (5mmx5mm) mounted on the front cylinder intake cam sprocket.

Semiseq/wasted spark sounds scary in my setup, it is a 60 degrees v twin, it would fire halfway during intake stroke on the second cylinder. There should be the option to avoid this kind of fallback.

Yesterday i drove for a while with single tooth cam and while enjoying the good job autotune was doing i lowered tps fuel load filter from 2% to 1%, then it leaned the very low openings around tps uncertainty and i started to experience more of those stumbles, so they may be more related to a fueling issue than a trigger one.

Not willing to be misleading, the difference from the twos is, in my situation, even with the non optimal trigger wheel shape, subtle.

VitorBoss, 2022-05-01:
From what I understood your engine have an even number of cylinders and can't use semi sequential / wasted spark, is that right?

gdiciocco, 2022-05-07:
Yes, it is a 60 degrees v twin and no, using semiseq/wasted will have disastrous effects.

VitorBoss, 2022-05-07:
> Yes, it is a 60 degrees v twin and no, using semiseq/wasted will have disastrous effects.

How many cylinders? It is easy to limit the changes to injection only, just need a bit.