speeduino/speeduino #836: ASE taper doesn't appear to taper over the duration

scudderfish, 2022-04-30:
This could be me making incorrect assumptions.  In the attached TS project running firmware 202204-dev, I've configured ASE to be a flat 50% for 10s and I'm running on a Mega2560 attached to a simulator.  I would expect the ASE to taper from 150% to 100% over the space of 10 seconds.  However it runs at 150% for 10 seconds and then tapers to 100% over 2 seconds.  This chop in the fuel is generating a fairly large stumble in my engine
[ASETaper.zip](https://github.com/noisymime/speeduino/files/8597071/ASETaper.zip)
.

pazi88, 2022-04-30:
If your ASE is set to 10s, it will keep the ASE enrichemnt for that 10s. So that's correct. What is your ase taper duration set?

scudderfish, 2022-04-30:
As suspected, problem exists between chair and keyboard :)

closed 2022-04-30 by scudderfish