speeduino/speeduino #841: BUG: Comms/startup issue Master branch

sbtoonz, 2022-05-05:
Processor: STM32F407VGT

Symptom:
When plugging board into power it fails to boot. 

Steps to replicate:
Compile master d2b9612409bff3b54a172f77014e4d6ca621db4e
Flash to stm32
unplug board 

plug board in 
board fails to boot speeduino

sbtoonz, 2022-05-05:
commenting out SDCard logging resolved this

most likely its searching SPI on boot and failing to find the sdcard logger

noisymime, 2022-05-05:
I will test this over the next few days, but issue is (at worst) limited to stm32 only. 
When you're seeing this, is the tune on the board (If there is one) set to use SD logging?

sbtoonz, 2022-05-05:
The tune was set to sd logger by default once loading the new INI into tuner studio (at least I am supposing so as this didn't exist in prior release version and that's what I was running prior to today) and today once I finally got connection made (by resetting multiple times) I could see the "Sdcard ready" green on the dash 

That led me to comment out the define for the sd logging in stm32 and attempt to repeat the connection error 

I could not repeat once commenting out

So this issue would present itself upon updating if the board was not provisioned for SD unless there's a way to default it false without undefining (I'm not sure if defines can be applied at runtime in this environment)

pazi88, 2022-05-05:
I can see this issue too. Most of the time it fails to boot if SD-card logging feature is compiled and even when the board boots, it takes really long time. I did not have this issue when I added the SD-card logging for STM32, so it's new thing. Need to investigate.

pazi88, 2022-05-05:
And yes if SD logging is commented out, the board boots up just fine. But I can't seem to find my ST-link dongle anywhere, so investigating this happens when I found it. Disabling SD-logging from TS-doesn't seem to help.

transi01, 2022-05-05:
where to comment out the SD logging?

pazi88, 2022-05-05:
> where to comment out the SD logging?

[bo](https://github.com/pazi88/speeduino/blob/dcb8fa009f6f99f2b88747cb6f45f653943145d2/speeduino/board_stm32_official.h#L37)

transi01, 2022-05-05:
Thanx :-)

pazi88, 2022-05-05:
Ok, I'm starting to get hang of this. It's actually the RTC that is causing the problem. If I have the RTC battery on the board (and RTC time/date set in TS) it boots up normally/immediately. If I don't have RTC battery on the board, it hangs up on boot or doesn't boot at all. This makes sense because the date/time is used for the log files and if there is no battery, the RTC isn't running on the board. Without SD-logging the RTC on STM32 wasn't used at all. This of course need to be fixed, because missing/empty RTC battery should not cause problems for booting up.

pazi88, 2022-05-25:
Yes problem is still there. I have the problem somewhat narrowed down. But no fix yet, although you can prevent it by adding rtc battery. If I don't come up fix before next release, josh needs to comment out that line in master.

pazi88, 2022-05-25:
> > Yes problem is still there. I have the problem somewhat narrowed down. But no fix yet, although you can prevent it by adding rtc battery. If I don't come up fix before next release, josh needs to comment out that line in master.
> 
> setting this define enables by default correct? This option can still be enabled via project settings in that use case if I am not mistaken?

Nope, it's build flag. SD-logging doesn't work at all if you disable it for building.

Disabling SD-logging in TS, doesn't really help this issue because the code does RTC/SD init anyways.

sbtoonz, 2022-05-26:
https://www.aliexpress.com/item/3256801471796407.html?spm=a2g0o.order_list.0.0.46d81802iJ8Wrx

just tried latest master with this board which has RTC crystal and SDCard slot provisioned 

Failed to boot

PR #858 (2022-06-27): Fix STM32 hanging on boot
In current master STM32 will hang on boot if no RTC battery is installed. The boot will either take long time or fails to boot completely. The reason for this is that the RTC will need to use external low speed clock (LSE_CLOCK) to be accurate. But "cold start" of the LSE_CLOCK takes up to 2 seconds and this is what causes the hanging. There seems to be no way around this long "cold start", so this PR basically moves the startup of the LSE_CLOCK to the moment of setting the RTC time. Otherwise internal low speed clock is used for RTC. Which is fast starting but not ac [... cut]
closed 2022-06-27 by noisymime