speeduino/speeduino #842: Feature Request: Fuel/Oil pressure in Bar

jackscr12, 2022-05-05:
I think it would be nice to have the ability to switch the unit of measurement for the fuel/oil  pressure.It's weird having a dash with all metric units and two gauges in imperial.
I tried to botch it by using a custom pressure sensor inputs  but that requires decimals otherwise the reading is off and when I tried that the value on the gauge  was behaving weirdly.

brunob45, 2022-05-05:
You could add a setting in the ini file, just like Celsius vs Fahrenheit
A good example is around line 348 in references/speeduino.ini

jackscr12, 2022-05-06:
Yeah I thought about Kpa but the problem is with a 7bar sensor the value needs to be 700kpa and that is over the 255 max limit .
Will check out the ini and see if I can cobble something up.
Thanks for the suggestions!

jackscr12, 2022-05-07:
Yeah so did not manage to do anything with the Ini as the problem is not just the unit of measurement but also the fact you can only enter whole numbers in the sensor range ... anyway this is way above my head .. so issue remains open for anyone interested and able to help.

Afroboltski, 2022-05-07:
I did the mods for this into a new branch of mine: [Metric-pressure-units](https://github.com/Afroboltski/speeduino/tree/Metric-pressure-units). 

_Side note: The mods only really need to affect fuel pressure and oil pressure, because for the fuel table there is a separate option in the `Options` -> `Preferred Units` menu, and for MAP you have 4 gauge options, kPa, bar, psi, or inHg(negative)/psi(positive)... actually technically the bar, psi, or inHg(negative)/psi(positive) gauges are MGP (manifold GAUGE pressure), not MAP, but I digress._

However, it is currently broken. I get the following warnings when switching between units:
````
5 Warnings:
Warning: MSQ Units Mismatch for fuelPressureMin! psi found in current configuration, kPa found in MSQ, values were not converted to new units.
Warning: MSQ Units Mismatch for fuelPressureMax! psi found in current configuration, kPa found in MSQ, values were not converted to new units.
Warning: MSQ Units Mismatch for oilPressureMin! psi found in current configuration, kPa found in MSQ, values were not converted to new units.
Warning: MSQ Units Mismatch for oilPressureMax! psi found in current configuration, kPa found in MSQ, values were not converted to new units.
Warning: MSQ Units Mismatch for oilPressureProtMins! psi found in current configuration, kPa found in MSQ, values were not converted to new units.
````
Seems obvious that this would happen at first glance, however the conversion works seamlessly for °C to °F conversions...

Let's see if anyone else has any ideas.