speeduino/speeduino #843: Acceleration Enrichment Taper not working?

ninjawarrior821, 2022-05-10:
I think my taper code is having issues. 
[image]
[image]

I've been noticing some herky jerkys on the highway and it seems the AE is randomly dumping a lot of fuel. I thought it was just a noisy TPS, but it shouldn't been adding that much with the taper.
I dug into the code a little (novice here) and looked at the taper which should be linear, but in actuality I only notice it turn off and on after the taper max.
[image]
[image]

I was wondering if anyone else had this issue. I'm also wondering if it's a similar issue to this fellow (integer vs float):https://forum.arduino.cc/t/calculating-percentages-yields-0-or-100-never-between/255106

Let me know if this is the complete wrong place to be asking for help, or if it's just a hardware issue.

noisymime, 2022-05-11:
Are you running on version `202202`? I suspect this is more likely related to the issue mentioned in #773 which has been resolved in master. 

The `percentage()` function in Speediuno does not use floats at all, so that thread mentioned is not relevant.

ninjawarrior821, 2022-05-11:
Fantastic, that looks like it could be it. I'll try loading the master and report back.

ninjawarrior821, 2022-05-12:
The car is running much better with the master! (I also CNCed an aluminum vTPS adapter to replace a 3D printed one)

[image]
I'll continue to log just to make sure everything is set. 
I assume what shows up in MegaLogViewer is at a slower sampling rate than what the board is running at. There appear to be spots where TPS_Dot = 0, but the AE still triggers...

Thanks so much!
I'm a huge fan of what y'all got going here with Speeduino.

felipegaspari, 2022-05-13:
Should this issue be closed or is it ongoing?

VitorBoss, 2022-05-13:
> The car is running much better with the master! (I also CNCed an aluminum vTPS adapter to replace a 3D printed one)
>
> [image] I'll continue to log just to make sure everything is set. I assume what [...]

The firmware samples TPS AE 30 times per second, the log depends a lot of your current frame limitation.
As the TPS resolution increased 2x the TPSdot is more sensitive to small changes, this is desired for alpha-N and short intake manifolds.

sbtoonz, 2022-05-24:
> Bad news. While the driving has improved, I've been looking at the logs and I believe the issue still remains. I did a quick test where I held the car ~2700rpm and stomped on the throttle. By my estimate the AE should be around 40% with a taper, but the logs show 100%. [image] The PW and other fuel logs also confirm a 100% was added...

I just checked some logs from driving around 2 days ago and my AE sits at 100% until the TPS variables are met then it will rise to around 140 and then back to flat 100%

sbtoonz, 2022-05-24:
[image]

noisymime, 2022-05-24:
Need to see some actual tune and log files here. There's not enough info to determine anything based on those comments alone

ninjawarrior821, 2022-05-24:
Here's the AE settings:
[image]

So by my calculations at a TPSdot of >1200, then 103*(3600-2750)/(3600-1400) = 40 (or 140 in the logs).

Here's my tune, look for log titled "Looking for AE Taper V3"
[MiataV3.zip](https://github.com/noisymime/speeduino/files/8764923/MiataV3.zip)
:

jackscr12, 2022-05-24:
Yep just checked my logs and there it is AE enrich going straight down no slope , and it is definitely  something in the code because it is the same tune as it was in april when I was running the last firm and AE was fine.

noisymime, 2022-05-25:
> Yep just checked my logs and there it is AE enrich going straight down no slope , and it is definitely something in the code because it is the same tune as it was in april when I was running the last firm and AE was fine.

That is normal, there's no taper to end AE. It just comes on, remains on for the 'Accel Time' duration and then turns off. 

The taper variables change the amount of AE that you get, not tapering it off over time

noisymime, 2022-05-25:
Have confirmed the issue here, both via logs and a unit test I've just put together. 

It's a different issue from the original one resolved above and causes the RPM based taper not to apply. Result is always 100% rather than the tapered amount. Not sure on the cause yet, but expect a fix incoming shortly.

closed 2022-05-25 by noisymime

ninjawarrior821, 2022-05-25:
> Have confirmed the issue here, both via logs and a unit test I've just put together.
>
> It's a different issue from the original one resolved above and causes the RPM based taper not to apply. Result [...]

My only thought is that the currentStatus.RPM might not be working. Stuck at 0 or something...

noisymime, 2022-05-25:
Fix has been pushed up. Please test if possible.

ninjawarrior821, 2022-05-25:
> Fix has been pushed up. Please test if possible.

Thanks! Will do!

sbtoonz, 2022-05-25:
> > @noisymime Just ran test on latest master my AE logs look identical to [9673dd8](https://github.com/noisymime/speeduino/commit/9673dd8fe62b423a602f1b0ffd90382cd29cb30f)
> > [2022-05-25_10.06.14AEpaTCH.mlg.zip](https://github.com/noisymime/speeduino/files/8772799/2022-05-25_10.06.14AEpaTCH.mlg.zip)
> > Would like to note in my prior logs the AE spike would happen substantially after TPS increase and in these logs it appears to happen 1:1 with my TPS increase
> > I also wanted to note that my AE still shows 100 but on the graph the line is at 0
> > Is 100 translating into 1.00 ? meaning X * 1.00 = Y?
> 
> I think you have the right idea. "100" on the graph should mean 100% of the VE table fuel, which is a multiplier of 1.00. The graph auto scales. so the bottom of the graph is always 100 because AE never takes fuel away.

OK then based on my TPS DOT logged my AE is working accurately.

I do know after pulling down these master changes I had to retune my VE table in spots

ninjawarrior821, 2022-05-26:
> Fix has been pushed up. Please test if possible.

It's now working as expected! 
[image]
The AE only reached 150 instead of 200, so the taper is working.

Thanks all!

commit e16108d8 (2022-05-25): Prevent overflow of the AE RPM taper. Fixes #843