speeduino/speeduino #847: Firmware locks up when loaded to Teensy 4.1

DetonationEMS, 2022-05-18:
USB connection will cause lockup and soft-brick of Teensy 4.1

(https://forum.micropython.org/viewtopic.php?t=12133) <-- Known bug with RTC on 4.1. May or may not be relevant. 

Firmware 202103 is the latest release that doesn't appear to have this issue.

noisymime, 2022-05-18:
From the testing I've done there are likely 2 issues impacting T4.1 at the moment. Will need to be careful not to make assumptions about which one is causing a particular issue. 

1. An issue that causes the entire T4.1 to hang on boot, likely related to the RTC. This was only from FW `202202` onwards and can be worked around by removing `SD_LOGGING` and `RTC_ENABLED` defines in `board_teensy41.h`
2. An issue that causes the Teensy loader utility to intermittently hang when uploading a firmware. The firmware will still boot and run fine however. This impacts FW versions from `202108` onwards. This appears to be related to some of the T4.1 timer configuration, but I haven't narrowed it down further than this.

transi01, 2022-05-20:
I have almost the same issue with Teensy 3.5.
Have deleted the entries for SD_LOGGING and RTC_ENABLED from Teensy3.5.h

Compiling did work afterwards and the LED on teensy whent lit again, so everything looks fine, but when I try to connect to Tunerstudio it hangs at 87% reading of the Tune and then it closes connection to Tunerstudio.

// Thomas

noisymime, 2022-05-20:
> Compiling did work afterwards and the LED on teensy whent lit again, so everything looks fine, but when I try to connect to Tunerstudio it hangs at 87% reading of the Tune and then it closes connection to Tunerstudio.

Whatever you're seeing is unrelated to this. What version of the fw is that with? 
I've burned and loaded over 100x T3.5 boards with the `202202` firmware and quite a few on the dev version without issue.

transi01, 2022-05-20:
well, before removing the the SD_LOGGING and RTC_Enabled lines compiling also did work, but LED on teensy did NOT lit and absolutely no connection to tunerstudio was possible.

I have done tests with many different FW dailys this one is from 16.05.22

transi01, 2022-05-20:
Hve NOW just for testing used the Speedyloader v1.5.1 (just for verification) with current Master and the result is a NOT lit LED on Teensy and impossible connection to the Board cause NO USB port will be recogniced by the pc .... so it seems not to be a problem just by my environment.

I have NO battery connected to teensy and I don't want to do so. But it shouldn't make a problem without Battery connected shouldn't it?!

pazi88, 2022-05-20:
> Hve NOW just for testing used the Speedyloader v1.5.1 (just for verification) with current Master and the result is a NOT lit LED on Teensy and impossible connection to the Board cause NO USB port [...]

Might be unrelated or not, but STM32 also has problems with RTC causing the board not boot. If there is battery installed and RTC set, all works fine. But if not, the board might not boot or boots really slowly. #841 It of course shouldn't be like that, but haven't yet figured out where to problem is excatly.

noisymime, 2022-05-20:
> Have NOW just for testing used the Speedyloader v1.5.1 (just for verification) with current Master and the result is a NOT lit LED on Teensy and impossible connection to the Board cause NO USB port will be recogniced by the pc .... so it seems not to be a problem just by my environment.
> 
> I have NO battery connected to teensy and I don't want to do so. But it shouldn't make a problem without Battery connected shouldn't it?!

What's your set with it? Do you have an SD card inserted, is it formatted etc? 
I would suggest doing an EEPROM clear and trying again.

I just grabbed a T3.5 and loaded the master from SpeedyLoader without issue. Connected to it in TS fine. Blanked the EEPROM and tried from scratch, also no issues.

transi01, 2022-05-21:
Have tested it more intensive and what I found out is:
After EEPROM clear programming using speedy loader did succeed and USB port of teensy was as solid as it should be :-)

Using my own compiler it was possible to program and it worked also after eeprom clear but when I remove and plug USB in again when Tunerstudio was connected after 2nd or 3rd time teensy's LED wont lit again and no connection could be granted again with TS.

When I now set frequency of teensy to 144 or 168 it works. All other frequency's fail as the 120Mhz does, very strange.
I have done research and it seems that teensy3.5 sets SPI speed internaly to 60Mhz for 120Mhz CPU clock and 56Mhz for 144Mhz and 168Mhz.

At which frequency do you compile teensy 3.5 firmware?
// Thomas

transi01, 2022-05-23:
Well, result of the overclocked Teensy 3.5 was that some input / outputs did no longer work as they should, so I digged in it more deep:

The result of the digging is:
The latest version of Arduino IDE and Teensyduino addon to work are:
Arduio IDE: 1.8.15
Teensyduino addon: 1.54
The problem is caused by the "core part" of the Teensyduino addon. When I copy the (Teensy3) core from version 1.54 over the 1.56 version the compiled built works again :-)

// Thomas

noisymime, 2022-06-01:
> At which frequency do you compile teensy 3.5 firmware?

All T3.5 build and testing is done at the stock 120Mhz. I'm still seeing T3.5 SD working fine with 1.56 though, no errors and no lockups.

noisymime, 2024-03-15:
I believe this can probably be closed. There have been a lot of changes over the last 2 years and in particular recently for the T4.1. 

I am not seeing any lockup issues currently on either the 3.5 or 4.1 boards, but please test and see if you are having any troubles with the current release.