speeduino/speeduino #848: Feature Request: 2nd VE table below TPS value

MykkH, 2022-05-21:
Feature Request. 2nd VE table active below specified TPS value. Would allow the use of 2nd VE as an idle only table as well as not get auto tuned by VEAL.

jackscr12, 2022-05-21:
Why would you need a whole VE table just for IDLE? and you can set the filters up in Auto tune to ignore the idle area ..so set filter to tune over 1500rpm and over 30kpa for example is a standard way to set up VEAL

digmorepaka, 2022-05-21:
Using the secondary table that's already there would solve this  though it seems it can only be used if a property is higher. I guess that would change this feature request to "allow use of secondary table with "lower than" conditional value" . It is rare for setups to need a separate VE table for the idle, but it does happen.

jankovalski, 2022-05-21:
@jackscr12 I understand the use of 2nd for idle only because that's what I'm doing right now, only with a button. The car is a classic 4 cyl NA. I have it tuned when it's warmed up. The goal is 14.7 AFR at idle and 12.6 at full power. And pretty much that's what I get with my current VE table. But when I start it cold, idle AFR is around 16. You'd say increase warmup enrichment. But when I do that my acceleration AFR drops down to some insane values like 10-11. I don't want that. The other solution that comes to mind is to increase VE cells around idle. Then I get 14.7 AFR when it's cold. But when it warms up, it becomes around 12. Way too low for idle. And I think using 2nd VE table just for this is a huge waste but I don't see another option for now. Much better solution would be to add a configurable ratio that would increase WUE for idle only. So for example WUE is 130% for the entire VE table but 150% for idle.

jackscr12, 2022-05-21:
I was genuinely curios why a VE table would be needed because I had not encountered this before. Did not mean to sound like I was criticizing it.

> @jackscr12 I understand the use of 2nd for idle only because that's what I'm doing right now, only with a button. The car is a classic 4 cyl NA. I have it tuned when it's warmed up. The goal is 14.7 [...]

Wait , can you not resolve this too rich acceleration mixture by tuning the AE cold Adjustment variables?   I had the opposite problem with AFR being very lean when accelerating when cold and had to increase the AE cold adjustment to 240% .

jankovalski, 2022-05-21:
> I was genuinely curios why a VE table would be needed because I had not encountered this before. Did not mean to sound like I was criticizing it.

All good. Different setups may require different solutions.

> Wait , can you not resolve this too rich acceleration mixture by tuning the AE cold Adjustment variables? I had the opposite problem with AFR being very lean when accelerating when cold and had to increase the AE cold adjustment to 240% .

I honestly didn't think of setting AE cold adjustment below 100% - gotta try that!

jankovalski, 2022-05-21:
[image]
I guess I can't do that...

jackscr12, 2022-05-21:
Your AE when hot is too high probably .. or some correction in your fuel calcs is off. I mean if you need to go sub 100% on AE when cold there has to be something off... why is there such a high discrepancy between cold and warm fuel req? perhaps the IAT fuel correction is wrong, or the injectors dead time

MykkH, 2022-05-21:
One wouldn't NEED an 16x16 VE for idle only. Even a separate idle fuel table like we get with ignition timing would be beneficial too. 

Most OE's have been using separate idle, part throttle and WOT VE tables for 40 years. It could be a useful tuning tool to isolate idle fuel from the main VE and having one more option for dialing things in.

jackscr12, 2022-05-21:
> One wouldn't NEED an 16x16 VE for idle only. Even a separate idle fuel table like we get with ignition timing would be beneficial too.
>
> Most OE's have been using separate idle, part throttle and WOT [...]

Yep more tables for each mode would make a more refined tuning experience for sure... and I would love to have multiple boost maps etc, but we are really struggling for space on the mega platform, as I am sure a dev will soon point out.
but maybe a small table for idle would be possible....

MykkH, 2022-05-21:
> 

...or just make the 2nd VE have "lower than" condition

Fera31, 2022-05-24:
go to SecundaryTables.ino    and change this
` if(currentStatus.TPS > configPage10.fuel2SwitchValue)`
 
 for this 
 ` if(currentStatus.TPS < configPage10.fuel2SwitchValue)`

MykkH, 2022-05-24:
Wouldn't there need to be a companion string in the firmware on the board to match the Greater than >, Lower than < switch in the tunerstudio .ino?

MykkH, 2022-05-24:
I'll change it and the tunerstudio speeduino.ini and give it a test. Thank You

MykkH, 2022-05-25:
Chaning the .ino to < and changing the .ini description of function has worked for me. I've also changed the .ino so the VE's work in 50rpm increments, making the 16x16 2nd VE a bit more useful as an idle table. Now if I can make VE values go to the 10th decimal I'll be all set.

poky, 2022-06-16:
> Chaning the .ino to < and changing the .ini description of function has worked for me. I've also changed the .ino so the VE's work in 50rpm increments, making the 16x16 2nd VE a bit more useful as an idle table. Now if I can make VE values go to the 10th decimal I'll be all set.

Hi, could you show how to modify the .ino file to work with 50rpm increments? Thanks!

HWright9, 2022-10-28:
Write any condition you like into an programable output. Set that output to an un-used digital pin on the mega (50 or so) and then set the 2nd table to trigger off that same pin. Works for me.