speeduino/speeduino #854: Rev Límit Reset.

randolreyes, 2022-05-26:
hello guys, I have done some projects both na and turbos, I dedicate myself to mechanics and tuning mostly in speeduino ecus and some hondas among others, of all the one I like the most is speeduino, that is why I am interested in improving even more so wonderful project, I must say that I know practically nothing about programming although I try, I would like to see if someone can help me with the implementation of something that I have in mind with the rev limiter.  Although this works very well in most engines, there are some in which it is practically impossible to make them cut slowly and moderately. This almost always happens with turbocharged inline 6-cylinder engines, at this point it occurs to me if In addition to the soft limit and the hard limit, a spark or fuel reset margin could be implemented, as some ECUs like Honda manage the cuts that normally occur between 200 RPM or 500 RPM, this request is motivated because I have observed that with the options available so far the speed of the cut is seriously affected by driving conditions and whether the vehicle is with heavy loads, medium loads or no load.  thanks for doing such a great job.

jackscr12, 2022-05-29:
I think you are talking about Hysteresis ? the amount the rpm should drop after initial cut than engaging the limiter again. So 200rpm hysteresis at 5000rpm would oscillate the rpm between 4800 and 5000.   Or do you mean some kind of hysteresis scalable with RPM and load , so for example 200 at 3000rpm  and 400 at 6000rpm?
with rolling cut hysteresis would be useless because the rpm in my experience never drops just slowly increases. 
Honestly the limiters in speeduino need more variables   , for example it would be nice to have rev limiter set to rolling and Boost cut set to FULL as the rolling does not reduce the boost but full cut on rpm limiter  is to aggressive.

pazi88, 2022-05-29:
Yes it sounds like hysteresis value and it's something on my to-do list, unless someone else does implement it first.

randolreyes, 2022-05-29:
Greetings guys, sorry for my bad English, I use a translator, if I refer to hysteresis since it is something that would help the limiter a lot by having the option of being able to lower the speed with which some motors make the cut, this would give a new approach in the The way turbocharged engines do it because when they produce a lot of power they go up in RPM very quickly and using the time delay to slow down a little this effect causes the mixture to start late and come out very hot towards the turbo exhaust valve and escape, I'm sorry I can't contribute from the code part with you but I'll be willing to help where I can.