speeduino/speeduino #856: Feature request, dynamic setting for TPS_READ_FREQUENCY

RempageR1, 2022-05-27:
Hi,

I`ve noticed the Acceleration Enrichment works much better for me with TPS_READ_FREQUENCY set to 15 instead of 30. This is with the latest code, so not related to #773. I would be happy to add a dynamic setting option to the .ini file and code, but before I investigate I would like to know it this is something that could make it into the main build.

Thank you.

pazi88, 2022-05-27:
That knid of setting doesn't make any sense. There isn't really need for setting like that, and if something, even faster update rate would be welcome. You just need to re-tune AE, because the 15Hz was so slow that it required some unusual settings for AE to work ok.

RempageR1, 2022-05-27:
For me, it did make much sense. It`s not the tune, it`s the TPSdot value that is very spiky at times or fails to pick up the slower throttle increases. Nothing you can do about that in the AE config. If the common consensus however is that it`s not needed, I`ll just modify it for myself whenever new release is available, that would also be fine.

closed 2022-06-03 by RempageR1