speeduino/speeduino #859: Hiccups during Autotune with NEW_COMMS

theKraid, 2022-06-01:
Both me and a friend have been experiencing some kind of hiccup or ignition cut at set time intervals (guesstimate every ~10 seconds) whenever Autotune is running. Presumably every time TunerStudio burns changes automatically?
It seems to be related to serial communication as the problem can be fixed by turning NEW_COMMS off in project properties.
Affected is both 202202 and current master, in 202108 the problem doesn't exist.
Our setups are both DIY-EFI's CORE4 PnP board in our MX5's.

noisymime, 2022-06-01:
When you're using auto tune, can you try with the `Update Controller` checkbox cleared as see if this makes a difference?

[image]

If you're compiling yourself on master, you can also try changing the value of `EEPROM_MAX_WRITE_BLOCK` in `storage.cpp` from 24 to say 12 and see if this makes a difference (I'm assuming you're running on a mega2560?)

If the above change does fix the issue, I'll just need to play with the tuning some more and figure out how aggressive I can go on those writes. The current value was chosen to go as fast as possible when loading a tune, but I hadn't actually tested it with auto-tune where it's writing every x seconds, so it might be a bit too aggressive.

noisymime, 2022-06-03:
Just pushed up a change that should resolve this one.

closed 2022-06-03 by noisymime

theKraid, 2022-06-03:
I could confirm that this was indeed a burn issue as it only happened with the box ticket. Didn't have enough time to recompile and find the breakpoint but you already implemented a fix 👍 Thanks!

commit 1d85100e (2022-06-03): Add dynamic EEPROM block write limit based on RPM. Should fix #859
commit ef39476f (2023-12-11): Adds support for toyota 3 tooth secondary and 2 tooth tertiary input (#1040)

* Disable SD on T4.1 temporarily to prevent lockups. See #847

* Fix unit test value

* Improve closedloop boost control (#736)

* First try on updating closedloop boost control

* added eeprom updates for backward comp[atibility

* fix merge conflict

* Change updates to next release EEPROM version.

* Fix some issues after merging master

* some cleanup to minimize changes and speedup review

* cleanup updates

* chances on updates to alaign with new table format

* move LUT to new config page in eeprom

* little c [... cut]
commit ec54ea8c (2024-02-02): Suzuki K6A Decoder (#1109)

* Disable SD on T4.1 temporarily to prevent lockups. See #847

* Fix unit test value

* Improve closedloop boost control (#736)

* First try on updating closedloop boost control

* added eeprom updates for backward comp[atibility

* fix merge conflict

* Change updates to next release EEPROM version.

* Fix some issues after merging master

* some cleanup to minimize changes and speedup review

* cleanup updates

* chances on updates to alaign with new table format

* move LUT to new config page in eeprom

* little cleanup

* Update storage.h comments

* fix some in [... cut]