speeduino/speeduino #863: TS_CommandButtonHandler

smb29, 2022-06-04:
TS_CommandButtonHandler
Can someone modify it, to have knock control from TS?

Autohome2, 2022-06-05:
What is it exactly you want this button to do?

smb29, 2022-06-05:
https://github.com/noisymime/speeduino/issues/764#issuecomment-1009306068

FazxFi, 2022-06-07:
Did you mean the knock setting window or something else

smb29, 2022-06-07:
I want something similar to the Knock settings window.
I think it would be a good thing for the community

pazi88, 2022-06-07:
Creating knock setting window to TS is not big task and to be honest, there is already knock settings for Speeduino, but it's hidden. Reason being that there is not really any knock sensing done in Speeduino code. I think it will come with speedyknock in future.

smb29, 2022-06-07:
well, if it's not a big deal, why don't you do it, instead of announcing it?
I know there are 241 mentions of Knock in the code, I even have some uncommented.
But, still, there is no trace in TS of Knock sensor. Not even on the programmable outputs

pazi88, 2022-06-07:
I mean for the TS GUI. In actual FW it's huge job. And yes it's mostly not done in the FW.

smb29, 2022-06-07:
I only ask that the TS GUI be modified

pazi88, 2022-06-07:
> I only ask that the TS GUI be modified

But you can't do anything with it, so what is the point?

smb29, 2022-06-07:
have at least the option in TS of Knock

pazi88, 2022-06-07:
> have at least the option in TS of Knock

Yes, but as is, it doesn't do anything. You first would need to code the features to the FW, to be able to use any of that.

smb29, 2022-06-07:
> > have at least the option in TS of Knock
> 
> Yes, but as is, it doesn't do anything. You first would need to code the features to the FW, to be able to use any of that.

so, everything is encoded in FW, (as far as Knock is concerned)
It does not even serve to assign an analog input in TS?

Just to clarify

smb29, 2022-06-07:
https://user-images.githubusercontent.com/62686180/148832992-983b5952-f7b2-41ec-a2dd-1d442a765278.png

How did this user do it?

pazi88, 2022-06-07:
> so, everything is encoded in FW, (as far as Knock is concerned) It does not even serve to assign an analog input in TS?
>
> Just to clarify

Nothing is really done in FW. Just some reservations for settings etc. you can see that gui TS, by uncommenting it in ini. But don't expect anything to work.

smb29, 2022-06-07:
Ok, 
just to finish and not bother more.
From where you have access to the graphical user interface, in TS?

Thank you for your patience passi

smb29, 2022-06-07:
good, very kind

mike501, 2022-06-07:
> It's a shame, it can't be done forever

Im sure it will be done once the hardware / firmware is ready for general
release.

Please remember, this isn't generally released yet so you are taking on
experimental software which may have defects. Any support you get is
because we're all nice people and not guaranteed, ideally you should be
prepared to support this yourself without anyone's assistance.

smb29, 2022-06-07:
Ok thanks I get it

closed 2022-06-07 by smb29