speeduino/speeduino #865: platformio.ini

smb29, 2022-06-05:
test_build_project_src = test_build_src

noisymime, 2022-06-05:
What's your actual issue?

`test_build_project_src` is a deprecated option now and hence has been removed. If you're having issues, upgrade your PlatformIO

closed 2022-06-05 by noisymime

smb29, 2022-06-05:
> What's your actual issue?
> 
> `test_build_project_src` is a deprecated option now and hence has been removed. If you're having issues, upgrade your PlatformIO

sorry, but it's in your code

noisymime, 2022-06-05:
Again, what's your actual issue???

smb29, 2022-06-05:
> Una vez más, ¿cuál es tu problema real?

does not matter leave it be. forget

noisymime, 2022-06-05:
> > What's your actual issue?
> > `test_build_project_src` is a deprecated option now and hence has been removed. If you're having issues, upgrade your PlatformIO
> 
> sorry, but it's in your code

The only place it’s used (That I can see) has been commented out: https://github.com/noisymime/speeduino/blob/1d85100ea708fc13c6d8e2efa3be2ce0b1d02d8c/platformio.ini#L18