speeduino/speeduino #867: Auxiliary inputs analog ports initialisation problem

dener4321, 2022-06-08:
Hello guys

It seems to me that the initialization of the analog ports when defined in the auxiliary inputs is not correct.
In sensors.ino file we have

```
byte pinNumber = (configPage9.Auxinpina[currentStatus.current_caninchannel]&127);
if( pinIsUsed(pinNumber) )
{
  //Do nothing here as the pin is already in use.
  BIT_SET(currentStatus.engineProtectStatus, PROTECT_IO_ERROR); //Tell user that there is problem by lighting up the I/O error indicator
}
else
{
  //Channel is active and analog
  pinMode( pinNumber, INPUT);
  //currentStatus.canin[14] = 33;  Dev test use only!
  auxIsEnabled = true;
} 
```
For me there are two problems here, first the byte variable pinNumber should be the same as the one used in the speeduino.ino file, since the field is defined as 0:5 or 6 bits, in this way
```
byte pinNumber = (configPage9.Auxinpina[currentStatus.current_caninchannel]&63);
```
And the other is that, for example, if the A0 port is defined in the tunerstudio, the value that will be saved in the speeduino will be 0 but the A0 port is defined in the Arduino library as 54, in this way the pin that will be defined as input will be digital 0.

Thanks, best regards

dener4321, 2022-06-09:
> Check PR #826

Okay, what about the other problem I mentioned.

> And the other is that, for example, if the A0 port is defined in the tunerstudio, the value that will be saved in the speeduino will be 0 but the A0 port is defined in the Arduino library as 54, in this way the pin that will be defined as input will be digital 0.

Do you have any knowledge if it's supposed to be like that? It doesn't seem right to me.

Corey-Harding, 2022-06-09:
Just note that PR has not been accepted yet.

I am not sure about your other question.  I am not a maintainer and have not been through the code in depth.  I am sure someone with more knowledge will chime in, the maintainer and top contributors here are very good.  Speeduino has a great community.

Afroboltski, 2022-07-09:
> And the other is that, for example, if the A0 port is defined in the tunerstudio, the value that will be saved in the speeduino will be 0 but the A0 port is defined in the Arduino library as 54, in this way the pin that will be defined as input will be digital 0.

Ahh ha. Yes.

Luckily this only applies to Auxilliary analog inputs, not analog inputs that are actually assigned to a function (e.g. ECT sensor, O2 sensor, etc.), because these "real" analog inputs go through this function:

```
byte pinTranslateAnalog(byte rawPin)
{
  byte outputPin = rawPin;
  switch(rawPin)
  {
    case 0: outputPin = A0; break;
    case 1: outputPin = A1; break;
    case 2: outputPin = A2; break;
    case 3: outputPin = A3; break;
    case 4: outputPin = A4; break;
    case 5: outputPin = A5; break;
    case 6: outputPin = A6; break;
    case 7: outputPin = A7; break;
    case 8: outputPin = A8; break;
    case 9: outputPin = A9; break;
    case 10: outputPin = A10; break;
    case 11: outputPin = A11; break;
    case 12: outputPin = A12; break;
    case 13: outputPin = A13; break;
  #if BOARD_MAX_ADC_PINS >= 14
      case 14: outputPin = A14; break;
    #endif
    #if BOARD_MAX_ADC_PINS >= 15
      case 15: outputPin = A15; break;
    #endif
    #if BOARD_MAX_ADC_PINS >= 16
      case 16: outputPin = A16; break;
    #endif
    #if BOARD_MAX_ADC_PINS >= 17
      case 17: outputPin = A17; break;
    #endif
    #if BOARD_MAX_ADC_PINS >= 18
      case 18: outputPin = A18; break;
    #endif
    #if BOARD_MAX_ADC_PINS >= 19
      case 19: outputPin = A19; break;
    #endif
    #if BOARD_MAX_ADC_PINS >= 20
      case 20: outputPin = A20; break;
    #endif
    #if BOARD_MAX_ADC_PINS >= 21
      case 21: outputPin = A21; break;
    #endif
    #if BOARD_MAX_ADC_PINS >= 22
      case 22: outputPin = A22; break;
    #endif
  }

  return outputPin;
}
```

See my fresh PR, #887.

noisymime, 2022-07-10:
Just to close the loop on this discussion from Discord:

> And the other is that, for example, if the A0 port is defined in the tunerstudio, the value that will be saved in the speeduino will be 0 but the A0 port is defined in the Arduino library as 54, in this way the pin that will be defined as input will be digital 0.

Whilst it's not formally documented, the `analogRead()` function will work  for both direct pin numbers and the analog ones. So `analogRead(0)` and `analogRead(A0)` will do the same thing (On a AVR anyway). 

See: https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/wiring_analog.c#L47

Hence why the code that's there already works OK. 

#887 Is good though as it makes this explicit and does address the init issue of using `&127` rather than `&63`

commit d527f173 (2022-07-10): Add necessary pin translations (#887). Fixes #867
closed 2022-07-10 by noisymime