speeduino/speeduino #885: Feature request: Oil pressure with standard 10-184ohm sender

ASMotionLab, 2022-07-06:
Hi,
I know oil/fuel pressure is already implemented, but they are tailored to transducers that output 0-5v signal.
I have a standard 'old school' 10-184ohm sender, and have built this resistor divider circuit to provide an analog voltage to a speedy pin:
[image]
My signal is 0.2v-2.45v from 0-145psi (0-10bar), and there is no way to configure this within TS.
[image]

My request is: adding a 2-point table that is un-greyed when you select 'custom' from the drop down, so you can specify voltages that correspond to 0 and 145 psi. And the ability to invert the signal too.

OR maybe a simpler solution, adding a '10-184 ohm sensor' option to the drop down, and having 2 boxes un-grey which are: "voltage at 0psi" and "voltage at 145psi".

I know it's easier just to roll with it and get one of the supported transducers but there are a lot of reported failures in reviews & comments, presumably from non-automotive transduders being sold. Proper motorsport grade ones are £100+ which is more than my speedy cost.

Thanks!

pazi88, 2022-07-06:
You should be able to juat extrapolate the pressure at 5v.

ASMotionLab, 2022-07-06:
@pazi88 tried that, but the max integer in the box is 255 and my pressure at 5v would be about 300.
Also my signal is inverted, so 'min' would have to be 300, which is also a 1 byte -127 to 127 range.

jackscr12, 2022-07-06:
why can't you make the voltage divider do 0-5v instead?

ASMotionLab, 2022-07-06:
> why can't you make the voltage divider do 0-5v instead?

Can do, the top resistor has to be decreased & that means there will be more current draw from the Speedy board's 5v regulator. I'm trying to limit it to <25ma to prevent any issues.
2.3v~ is way more than enough resolution for me! I think that's like 500 steps or so

ASMotionLab, 2022-07-07:
@jackscr12 that's possible for sure, but still doesn't address the inversion 😣
I might just use a tiny Arduino to translate the signal and emulate a supported transponder but it seems kinda crazy to  have to do that 🤣

jackscr12, 2022-07-07:
> @jackscr12 that's possible for sure, but still doesn't address the inversion 😣 I might just use a tiny Arduino to translate the signal and emulate a supported transponder but it seems kinda crazy to have to do that 🤣

yeah an extra arduino would be a bit silly, I already have one for the EPAS... I'm in the same boat as you but I didn't get around to tackilling the issue... I have a sender for my analogue gauge and I would prefer not to trust my engine to a crappy ebay sensor and more fittings that can leak....
So please code Gods bless us with a mod :)

ASMotionLab, 2022-07-07:
@Fera31 I'd happily use OEM but I'm not aware of any 3 wire OEM transducers, is there a list somewhere of these? I thought all OEM are usually 1/2 pin potentiometer-to-ground type.
Had another look and Turbozentrum stock what look like legit transducers for 60£-80£, so i might just bite the bullet.
Still, support for the VDO types with voltage divider would be super cool 👌

Fera31, 2022-07-07:
> @Fera31 I'd happily use OEM but I'm not aware of any 3 wire OEM transducers, is there a list somewhere of these? I thought all OEM are usually 1/2 pin potentiometer-to-ground/VDO type, what I'm trying to get to work. Had another look and Turbozentrum stock what look like legit transducers for 60£-80£, so i might just bite the bullet. Still, support for the VDO types with voltage divider would be super cool ok_hand

the list is very extensive.
  but you can use of
GM 12616646 D1846A
VW 06E906051K
PSA 9802152780 1131K8
there are many more examples.
common rail sensors also work but they are very large in terms of measurements
  300 or 500 bar
i have gm style sensors and work great
[image]
[image]

ASMotionLab, 2022-07-10:
Well hot diggidy dang it works.
I modified the circuit as suggested by @noisymime:
[image]

and did some sketchy guesstimations in excel
[image]

and came up with these values:
[image]

and it's working well
[image]

need to hook up a mechanical gauge to verify the accuracy, but for now I'm happy.

It would still be a nice to have feature in software, as the resistor divider method puts the values right on the edge of the input range and adds more current drain on the built in voltage regulator.

@Fera31 thanks for the list and the nice install images. I had a look at most likely will go for one of the sensors in the list. I decided against it for now as I'd need to buy a separate pigtail, get a thread adapter & run more wires through the bulkhead. But definitely will switch in the future 👍

ASMotionLab, 2022-07-12:
Small update:
The resistor values maxed out the config values in TunerStudio, and with engine-off it was reporting 6psi or so.
So I changed the bottom resistor from 30 ohm to 22. Current consumption is up slightly, but now the values can be scaled down a little, leaving some wiggle room for calibration in software.
[image]

[image]

closed 2024-09-26 by noisymime