speeduino/speeduino #898: Rolling cut isnt cutting it

runesmdk, 2022-07-22:
Pun intended. 

Rolling cut of just 1 event per cycle is too little.

On a 4 cylinder, it should happen over a total of 5 ignition events, 4 should cut, 1 fire, reset and repeat.

On a 6cyl, 7 events with 6 cut and 1 fire, and so forth.

That, or atleast half, like 4 cyl, 3events with 2 cut 1 fire.

That way it will keep going through the cylinders and not flood some when in the cut area. 

Rolling now barely works even with very heavy retardation

brunob45, 2022-07-23:
On my project, I've implemented a gradual rolling cut for launch control, flat shift and regular limiter.

I found this algorithm on Motec's forum https://www.motec.com.au/forum/viewtopic.php?f=11&t=175&hilit=randomiser

This video makes a great job at explaining how a professional rpm limiter works (especially the diagram at the 20 minute mark) 
https://youtu.be/Au-7kYO4sak

Maybe this is something worth looking into for speeduino also.

noisymime, 2024-03-15:
Closing as the new random rolling cut limiter introduced in 202310 should resolve this. The HP Academy video posted above was used as a reference for this implementation. 

The only addition worth considering here would be the 'seamless fractional prime cut' mentioned on the Motec page that prevents the same cylinder from being cut consecutively. Currently there is no consideration of this and it might help lower cylinder count engines (Particularly ones with paired/wasted arrangements). 
I'll have a play with this feature.

closed 2024-03-15 by noisymime