speeduino/speeduino #905: 202207 stepper idle closed loop hunting and adding steps as rpm goes up.

wjones1972, 2022-07-30:
I installed 202207 and have issues with Closed loop Idle. In testing and open loop higher step count increases rpm and lower slows the engine down and open loop works well. However when I try closed loop it is unpredictable and mostly tries to run away. you can watch the step count going up as engine rpm's go up but it isn't 1to1 the step count is erratic. I rolled the firmware back to 202108 and it works as expected. So I am pretty sure there is some kind of issue in 202207. If there is more I can share I am happy to. I am also happy to help any way I can. testing beta firmware or whatever.

ric355, 2022-07-30:
Will probably help investigations if you post your tune.

VitorBoss, 2022-07-30:
I recommend you use the mixed(OL+CL) control for PWM and Steper, is miles better than pure CL.

I have been using it even before the PR got merged and is very stable, closed loop isn't and never will be