speeduino/speeduino #907: Idle control no PWM output. Single wire, open PWM STM32

NickZ1969, 2022-07-31:
Not sure if it effects the other processors.
I SOMETIMES get it to function by changing to Stepper, reboot, change back to PWM single wire idle control.
Change back to 202108 and everything works perfect again.
Im using the same pin mapping for both Firmwares, Same hardware and Same car. I have also used the scope on the output.

NickZ1969, 2022-07-31:
i have found what makes it work, i had nothing else PWM turned on, once i enabled Boost, both the boost and PWM idle control worked, But the idle control wont activate until about 2000rpm and then functions normally. If i disable boost control and reboot Idle control doesn't work again.
but enabling boost control and idle works again.
might be the same for VVT, i didn't experiment with it as i wouldn't have thought idle would have needed it. so there is something wrong with the PWM

NickZ1969, 2022-07-31:
Tjeerd has fix it in the board_stm32_official.h.
Please replace line 269 in board_stm32_official.h with:
#define IDLE_TIMER_ENABLE()  (TIM1)->SR = ~TIM_FLAG_CC4; (TIM1)->DIER |= TIM_DIER_CC4IE; (TIM1)->CR1 |= TIM_CR1_CEN;

Tjeerdie, 2022-08-01:
PR #908 Is the fix for this

commit 827f078e (2022-08-02): Fix for idle PWM on STM32. Enable idle timer when idle pwm is enabled. Fixes #907
commit b1ce0cad (2022-08-02): Fix for idle PWM on STM32. Enable idle timer when idle pwm is enabled. Fixes #907
closed 2022-08-02 by noisymime