speeduino/speeduino #911: Speedyloader not updating signature in Speeduino

mppickett, 2022-08-10:
I came across a small problem where it looks like Speedyloader doesn't update (always?) the signature in the arduino. In my case, I upgraded to the 202207 firmware using Speedloader and have been using it with Tunerstudio for several weeks with no problem. I got a new phone and reinstalled MSDroid. MSDroid wouldn't connect and complained that the firmware versions didn't match. It said the firmware signature in the Speeduino was "speeduino 202204-dev. I reconnected using Tunerstudio and got a similar error. I repeated the connection to Tunerstudio after shutting down everything and was successful at connecting with no errors.

I serial connected directly to the arduino and executed the "S" command and it responded with "Speeduino 202204-dev." Somehow that signature had persisted after the last Speedyload.

I downloaded the 202207 files, compiled and uploaded them to the Speeduino using the Arduino IDE. That corrected the signature reported by arduino and the MSDroid connected after that point.

Probably an obscure problem, but doing an IDE compile and upload is a workaround.

NickZ1969, 2022-08-16:
This comes up if you haven't used the correct INI or haven't adjusted the settings in project properties. Its using the legacy serial comms settings that still have the old signature.
If the correct ini or the settings fixed (Controller type and New comms enabled) in the project settings this will work correctly. these are not carried over from the old tune and have been recently added.

mppickett, 2022-08-16:
Thanks, Nick. I don't think that's the case, in this example.

 SpeedyLoader (V1.5.1 Windows) always sets the signature in the firmware on the board to "2022.04-dev" and TunerStudio New Coms ignores it if the INI file matches the Project setting. However, the old signature persists on the firmware unless you do a direct IDE load of the firmware.  Not a problem with TunerStudio, but when you use MSDroid on your smartphone to do data logging, it can confuse TunerStudio and sometimes MSDroid and cause an error.

I took my backup V0.4 board and did a series of signature reads and firmware uploads and was able to reliably replicate the problem. I'll upload the screenshots, but the upshot is that the current release of SpeedyLoader always sets the signature to "2022.04-dev." IDE uploads always correctly set the firmware signature. Not a big problem, but maybe it can be touched in the next release of SpeedyLoader.

Signature Before 1st SpeedyLoader
[image]

Speedyloader Screen
[image]

Signature After 1st SpeedyLoader
[image]

TunerStudio Board Version Setting
[image]

TunerStudio New Coms Setting
[image]

TunerStudio Loading a 202207.ini Tune
[image]

TunerStudio Project Signature After Burning Tune
[image]

Firmware Signature After TunerStudio Burning Tune
[image]

Firmware Compile and Upload Using IDE
[image]

Firmware Signature After IDE Upload
[image]

Signature After 2nd SpeedyLoader Upload
[image]

Signature After 2nd IDE Upload
[image]

NickZ1969, 2022-08-16:
Strange you are getting this, Im using speedyloader and uploading about 5 ECU's a day but not having your issue.
Speedyloader doesn't change the code, it just downloads a precompiled bin file.
[image]
[image]
If we look at the code in Legacy you will see the signature it's displaying when you are requesting it in you examples.

NickZ1969, 2022-08-16:
I think i know your issue. did you use speedyloader when the new firmware was released, The very first one had the wrong signature and was updated shortly after with the new signature, so if you use speedyloader it will keep burning the BIN file you originally had and doesn't download a new one each time. you need to locate the Speedyloader directory and delete the 202207 bin file you have and then restart speedyloader and it should download the correct bin with the fixed signatures, you IDE burned version would be the fixed version and working correctly.

mppickett, 2022-08-16:
Bingo. On Windows SpeedyLoader puts the binary file in the <username>/download directory with a .hex extension. As you noted, it doesn't redownload it everytime. You can force it to download by deleting the file. In my case, I deleted the 202207.hex file and when I reran Speedyloader it redownloaded 202207.hex with the correct signature. Problem solved.

Many thanks for the sleuthing!

noisymime, 2022-10-03:
Closing as this was a temporary issue. 

If anyone else runs across this, delete the *.hex files from your Downloads directory

closed 2022-10-03 by noisymime