speeduino/speeduino #921: [Suggestion] Conditional compiling and code refactor

smclt30p, 2022-08-31:
Hi,

I looked at the code base, and saw room for improvement. I have a few points.

1) Currently the code memory on the 2560 is running quiet high, and we can't integrate many features in the future due to this.
I suggest adding conditional compiling using the preprocessor, using a config.h header (Autoconf style). This way, we can compile out
certain features such as WMI/Nitrous that one may not use, but frees space for additional features like LPG, Can dashboards etc.
Also, a warning would be added to TunerStudio that says for example: "This feature is not compiled on this board".

2) Also, looking at the code there is a very messy implementation for outputing, that can be refactored greatly.
For example, the file _scheduledIO_ is full of functions that are used for opening and closing of the injectors and coils named for example _beginCoil2and4Charge_. This can be refactored to beginCoilCharge(COIL2, COIL4). 
Is there a reason why there is a _huge_ list of functions calling each other? I would like to refactor a lot of this.
Also, each injection/ignition event does an if/else check that is inline, consuming **huge** amount of memory. This can be refactored greatly.

I would like to do these refactorings/changes, and I wanted to just consult before making these changes.

noisymime, 2022-08-31:
> This way, we can compile out certain features such as WMI/Nitrous that one may not use, but frees space for additional features like LPG, Can dashboards etc.

The vast, VAST majority of Speeduino users don't compile the code for themselves, think maybe 1 in every 50 users who do.  So whilst I'm not against it, it has to be done in a way that avoids confusion for the other 49 out of 50 users. 

You will need to have separate ini files based on what is compiled in too as the memory needs to be mapped accordingly in it. Will be slightly painful to maintain. 

>  injectors and coils named for example beginCoil2and4Charge

These are used as callback functions. The scheduler doesn't know (nor should it) whether then engine is sequential, semi-seq, what the coil wiring is etc, it just takes a callback function that does what it needs to at the relevant time. So these functions are there simply as convenience callbacks. 

> Also, each injection/ignition event does an if/else check that is inline, consuming huge amount of memory

Taking a large amount of EEPROM space, NOT memory. The inline inclusion of this code prevents a jump out of the ISR, which avoids a significant performance hit.

noisymime, 2024-03-14:
Closing as either complete (Schedule refactoring) or not going to be added (Conditional feature compiles)

closed 2024-03-14 by noisymime