speeduino/speeduino #932: 8 cyl basic distributor doesn't work.

cavitating409, 2022-09-26:
if you setup as basic distributor and 8 cyl. it only outputs ignition pulses on ign 1, for a few seconds, then goes to outputting 4 pulses in a row, then skips 4 pulses then repeats continuously. played with filtering, pos or neg input and output just to test. no change. change to any other number of cyl's and unit works perfectly. just obviously calculates degrees and rpm incorrectly.  currently it runs great using 4 cyl. and just doing math in my head for rpm and degrees. FYI its 202207 on ua4c. high speed logger shows perfect trigger input on any cyl.

jonbills, 2022-09-26:
using 'spark output mode' == single channel, right? 
after a quick look, I have a feeling that init.ino line 806/7 and maxignoutputs=1 and/or MAX_CRANK_ANGLE=90 isn't compatible with the ignition scheduling code at speeduino.ino line 1098

cavitating409, 2022-09-26:
Yes single

noisymime, 2022-09-27:
I think this is likely related to the dwell limit calculations. It only begins to occur once the dwell+spark time approaches the limit (Based on the current revolution time). 

At a quick look in this code, it doesn't appear to be taking the spark duration into account in one of the calculations when `Single Channel` is selected as the spark mode. Will continue investigating

noisymime, 2022-09-27:
Can you post your tune please

noisymime, 2022-09-27:
Can you please try with the latest master code

cavitating409, 2022-09-27:
Yes I'll post tune. Not sure exactly what settings are in there now. I did
move dwell down to under 1ms and maxed it out. No change, but it also does
do dwell backwards so it doesn't work, clue? I need rising trigger and when
I move dwell it changes the time the line is high which is while it's
sparking not charging. I'm triggering an external amp so it doesn't effect
me. But figured I'd mention it. I did also change to falling and dwell
reverses. But doesn't fix fire 4, skip 4 issue.

noisymime, 2022-09-27:
Did you have a chance to test with the latest code on master? Assuming it’s the same issue, it should be resolved now.

cavitating409, 2022-09-27:
Haven't been near it. What do you mean by on master?

noisymime, 2022-09-27:
I pushed a change up on the master (aka development) code that I believe should resolve this. If you want to try it, scroll to the bottom of the firmware list in SpeedyLoader and select the `master` firmware and load that.

cavitating409, 2022-09-28:
Will do. Thank you

foppert, 2022-09-28:
I don't know if its helpful, I found for my V8 single coil setup I had to run with "new ignition mode" disabled. If I had "new ignition mode" enabled the car would idle ok but any revs would result in backfiring and clear timing issues.

I'll try this updated code when I can and see if I can re-enable "new ignition mode".

noisymime, 2022-10-12:
Any feedback on whether this change has made a difference?

cavitating409, 2022-10-12:
sorry i got so busy havent had a chance to test it.

foppert, 2022-12-27:
Just tested newest firmware, I am still experiencing issues with new ignition mode.