speeduino/speeduino #955: Fuel Injector Timing has wrong description

HWright9, 2022-10-28:
Fuel injector close angles are actually start angles as implemented in the firmware.

I think it would be better if they really were close angles since this would give better timing to intake valve opening with variable pulse widths, but this is not how it's implemented now. To fix this the fuel scheduler should swap the start and the end times, similar to how ignition angles are done. Alternately if start angles are desired, change the .ini to reflect this.

[image]

Traceback through the sofware:

[image]
calculated from:
[image]
calculated from:
[image]
calculated from:
[image]

noisymime, 2022-10-28:
Have you verified this on a scope? I believe the current implementation is correct. 

You have this line:
`unsigned int PWdivTimerPerDegree = div(currentStatus.PW1, timePerDegree).quot; //How many crank degrees the calculated PW will take at the current speed`

And then within the `calculateInjectorStartAngle()` function you have:
`tempInjectorStartAngle -= PWdivTimerPerDegree;`

That line is taking the lookup from the injector angle 2d table and subtracting the number of degrees that the PW goes for. The result is the start angle.

HWright9, 2022-10-28:
Ok I missed that. tricky.

closed 2022-10-28 by HWright9