speeduino/speeduino #957: TunerStudio sends wrong ADC scaling for sensor calibration values

DeionSi, 2022-10-28:
I think I found a problem with TunerStudio. The ADC scaling of the sensor calibration tables in Speeduino goes from 0 to 992 instead of 0 to 1023. Looking TunerStudios std_ms2gentherm.log the Volts scaling goes from 0 to 4.844 instead of 0 to 5.

[std_ms2gentherm.log](https://github.com/noisymime/speeduino/files/9891796/std_ms2gentherm.log)

Coolant output of "z" command to speeduino (prints calibration values).
0, 216
32, 216
64, 185
96, 166
128, 153
160, 143
192, 136
224, 129
256, 123
288, 118
320, 113
352, 109
384, 105
416, 101
448, 97
480, 93
512, 90
544, 87
576, 83
608, 80
640, 76
672, 73
704, 70
736, 66
768, 62
800, 57
832, 52
864, 47
896, 41
928, 33
960, 21
992, 0

At lower temperatures this causes a noticeable temperature difference. In my case:
17C -> 12C
1C -> -7C
-7C -> -19C
-20C -> -40C

Do we have any direct contacts at TunerStudio this can be reported to?

DeionSi, 2022-10-28:
Actually, it seems the issue is in Speeduino. I'm halfway done with the fix.

pazi88, 2022-10-29:
Whst speeduino version you are working on. Master uses 32 values. Not 1024.

DeionSi, 2022-10-29:
Yes, it uses only 32 values. But those values are scaled from 0 to 992 but should be scaled from 0 to 1023.

PR #959 (2022-11-30): Scale IAT and CLT calibration to ADC 1023 instead of 992
Fixes #957

The calibrations for IAT and CLT are currently scaled for ADC values from 0 to 992. This causes a bigger temperature mismatch the lower the temperature gets. This pull request changes it so the full range of 0 to 1023 is used.

Calibrations values before and after.
ADC old | ADC new | Calibration | Actual temp
-- | -- | -- | --
0 | 0 | 122 | 82
32 | 33 | 202 | 162
64 | 66 | 169 | 129
96 | 99 | 151 | 111
128 | 132 | 140 | 100
160 | 165 | 130 | 90
192 | 198 | 123 | 83
224 | 231 | 117 | 77
256 | 264 | 111 | 71
288 | 297 | 106 |  [... cut]
closed 2022-11-30 by noisymime