speeduino/speeduino #966: Any way to get decimal placess VE? Need more precision to adjust mixture in idle

chaoschris, 2022-11-19:
Hi guys,
I was surprised nobody had this issue until now or I haven't found it in the forum.
Is there any chance to get one decimal place in the VE table? This must not be over the whole table, only for idle or any extra function/table would be enough.
We have a use case in a car with big injectors. On idle VE is about 21, WOT is round 140. Tuning with integer values isn't precise enough. Changing the value by 1 digit the mixture is either way too lean or rich in idle. We are talking about 5 percent more or less fuel, which is really a lot.
Or is there any way to handle this issue we haven't seen until today?
Best regards,
Chris

VitorBoss, 2022-11-19:
The only hack available is to half your required fuel value to use double values on VE, this will achieve same as 0.5% precision.
Unfortunately is not possible to change the table resolution at this moment.

21 VE seems like your injector dead time is wrong.

pazi88, 2022-11-19:
Is the multiply map on? It will increase idle area ve resolution so much that it isn't problem annymore.

Tjeerdie, 2022-11-19:
As @pazi88 said. I think your did not turn **¨Multiply VE value by MAP¨** to **¨fixed"**. This setting is below the VE table in one of the 3 settings. To get the VE to that is tuned with the option **"off"** to work with that setting to **"fixed"** you should multiply each map row of the VE table by 100/"kpa value row". 

That is basically the opposite of that the speeduino code does. This should make your map more flat and all values somewhat close to 100.

PSIG-dev, 2022-11-23:
+1, as stated in the Forums, a common solution for this effect is scaling.  Scaling is used to bring-up the operating range to higher numbers, e.g., VE values * 1.5 provides a new range of 31.5 (round) to 210.  The Req_fuel multiplier is likewise divided by the same factor so fueling is equivalent.    This still leaves room for additives (warmup, AE, etc) at the top, while giving the effect of control resolution (finer steps of VE) at idle.  

This will not save you if the injectors are operating in non-linear regions.  Altering fuel pressure, using 1:1 regulation (if currently using fixed pressure), or using rising-rate pressure regulation may help in that case, either increasing low-MAP idle PW, or reducing high-MAP PW, or both.