speeduino/speeduino #968: Feature request - TACHO ActiveLow or ActiveHigh

serifis84, 2022-12-04:
Is there any chance to have this feature on the newer releases ?  I would like to have the ability to select in the Tuner Studio whether tacho pulse will be active_low or active_high.

digmorepaka, 2022-12-04:
Can't comment on the software implementation but you can hook up an NPN transistor to the output+ground+pullup to invert it.

serifis84, 2022-12-05:
> Can't comment on the software implementation but you can hook up an NPN transistor to the output+ground+pullup to invert it.

Hello mate,

Some of my projects requires connecting a 2560 board to OEM ecus (current project using jump wires to components on Magneti Marelli IAW AP 41). They are already having inverters and drivers on-board. Spark settings, programmable output settings, other inputs like launch etc have the ability to invert signals.