speeduino/speeduino #981: Duplicate idleLoad entries in Serial3/CAN data

iltHeo84, 2023-01-12:
Is there a specific reason for bytes 37 and 90 of the live data to carry the exact same value, as per **cancomms.ino** code?

`fullStatus[37] = currentStatus.idleLoad;`
`fullStatus[90] = currentStatus.idleLoad;`

Thank you!

Autohome2, 2023-01-12:
It looks like a typo has crept in at some point.
90 should be currentstatus.idleduty.

iltHeo84, 2023-01-12:
> It looks like a typo has crept in at some point. 90 should be currentstatus.idleduty.

currentstatus.idleDuty has been removed on March 2022 with Pull #806.

maybe 90 could be used for idleup correction? maybe the sum of idleup %/steps if active and airconidleup %/steps that is added to idleLoad

Autohome2, 2023-01-12:
Yes it could be re used for another function as long as the new data fits into the existing single byte of data.