speeduino/speeduino #983: Possible bug?

AshtonRey, 2023-01-15:
Good morning, I’ve been chasing a on going issue with my 1991 Mazda Miata 1.6L. I’ve been trying to set up an IACV on my Miata as I have ITBs. I’ve been having a strange issue where I cannot get tunerstudio to communicate with the IACV while it’s running. I would change the DC % up and down and adjust the Frequency but with no luck. I’ve checked all my wiring many times, used multiple different similar IACV that I know work and checked for any vacuum leaks but it all checks out. Ive also gone a re uploaded the firmware to the ECU and took a log and shows my changes when adjusting the DC%. I’ve hooked up a multimeter and saw DC voltage go up and down in sync with my changes on tunerstudio to the DC%. I will say i have full control/ the IACV responses to my changes when the car is in ACC mode since I have the option on for cranking to help with starts. But as soon as the engine is running I cannot get it to do anything. I’m stumped, any help or pointers would be appreciated!

jonbills, 2023-01-15:
Sounds like two functions trying to use the same pin. Can you attach your tune and a log?

AshtonRey, 2023-01-15:
@jonbills it would not let me attach the files in GitHub, so I made a temporary link. Thank you for taking a look! 
Link: https://file.io/z2aIH77901mj

jonbills, 2023-01-15:
I don't see any pin conflicts. I'm not sure I understand the problem description on second reading. What's the problem again? are you not seeing something in Tunerstudio that you're expecting?  Tunerstudio doesn't communicate with the IACV - Speeduino does, and tells TS what it did.  The log does show IACV DC changing.

AshtonRey, 2023-01-15:
@jonbills Sorry I meant to say speeduino not tunerstudio. But my issue was even though I’m making changes to the IACV DC the physical IACV valve doesn’t move/change how much air gets past it when I make DC% changes. I’ve swapped different IACVs, Re wired and checked wiring, vacuum leaks, and saw voltage reaching the IACV connector and add and take voltage away as I moved the IACV DC%.

jonbills, 2023-01-15:
how are you measuring the air getting through the IAC? for my info, how is the idle air shared between the ITBs?
might be a good idea to ask this question in the help channel in discord.

AshtonRey, 2023-01-15:
I’ll ask in discord help chat! I appreciate it!

closed 2023-01-15 by AshtonRey