speeduino/speeduino #987: Feature request: MAP/BARO as load

iltHeo84, 2023-01-19:
When I started tuning my engine (N/A), I turned on the "Multiply MAP by Baro" option in the VE table, (wrongly) assuming this would scale the MAP reading used as the load source by the current reading of the Baro sensor.

In fact, if I read the code correctly, it only affects PW by means of multiplying the current VE value for the MAP/Baro ratio, but the load source used to read the VE table is not changed.

Since in my area (around 200m above sea level) the barometric pressure is in the 94-97 range, it prevents VEAL to precisely tune the 100 kPa row.

This request is for an option (just like MAP, TPS, IMAP/EMAP) to use MAP/BARO as the load source for the VE table.

pazi88, 2023-01-19:
How come the map/baro load helps you tune the 100kpa line?

iltHeo84, 2023-01-19:
> How come the map/baro load helps you tune the 100kpa line?

Assume Baro = 95kPa.

At WOT, a N/A engine will see a manifold pressure of the same 95 kPa, thus Load = 95

But if you factor in the Baro value,
Load = MAP/Baro*100 = 100

pazi88, 2023-01-19:
Ok. But in that case you would be hitting 100kPa VE value both 100kPa and 95kPa baro readings. Which yields to same amount of fuel on both cases, which isn't correct. You would need less fuel in case of the 95kPa baro situation, because less air gets into engine.

iltHeo84, 2023-01-19:
> Ok. But in that case you would be hitting 100kPa VE value both 100kPa and 95kPa baro readings. Which yields to same amount of fuel on both cases, which isn't correct. You would need less fuel in case of the 95kPa baro situation, because less air gets into engine.

Correct. That is what the Barometric Correction table would take care of.
Just for reference, MS has an equivalent Load option, "Percent Baro".

pazi88, 2023-01-19:
But you can use the baro correction just fine even without map/baro load. And in speed density mode that curve isn't really even needed. Run the fixed map multiply mode (not the multiply map by baro) and use the baro correction curve if needed.

There has been lot of discussion in discord about this and you can join it. But overall consensus have been that the multiply map by baro isn't correct way to do it. The fixed 100kpa multiply is the more correct way.

pazi88, 2023-01-19:
Nope. But you really can't tune it without the trip to seaside.

iltHeo84, 2023-01-19:
> Nope. But you really can't tune it without the trip to seaside.

Ok, but one could tune the WHOLE VE table at his usual baro values and then only adjust the Baro Corr table at different altitudes (just like you do with IAT Density)

VitorBoss, 2023-01-19:
If I understood correctly you want to change the load axis to a percentage, so you can tune it up to 100%, in that case it make sense. But what you are saying is to be able to tune it up to 100kPa, that would need you to drive in sea level, kpa are absolute value, you can't use 100kpa axis on a 94kpa reference

iltHeo84, 2023-01-19:
> If I understood correctly you want to change the load axis to a percentage, so you can tune it up to 100%, in that case it make sense. But what you are saying is to be able to tune it up to 100kPa, [...]

The option I am suggesting will enable anyone at any altitude to tune the whole table up to 100kPa, making the vertical axis a proper "load" axis (if MAP is 95kPa and baro is 95kPa, you are actually loading the engine as much as you can at that altitude).
Just like to tune it at a given air temperature and later adjust IAT Density, you can tune it at 95kPa baro and later adjust Baro Correction

iltHeo84, 2023-01-19:
> Again tune a 100kpa row with 95kpa doesn't make any sense, until you discover why is the case you won't get a different answer.

It doesn't make sense if you consider the VE table as the "real" volumetric efficiency of the engine.
In reality, it is just another lookup table, as all the tables in speeduino/tunerstudio.
If my MAP is 95kpa and baro is 95, my engine is working at 100%load AT THAT SPECIFIC BARO VALUE.
Will it need the same amount of fuel at 100% load at 100kpa baro? Obviously not, but the efficiency of the engine did not change.
Just like IAT, it is an external variable.

Benjamin4456, 2023-01-19:
> It doesn't make sense if you consider the VE table as the "real" volumetric efficiency of the engine. In reality, it is just another lookup table, as all the tables in speeduino/tunerstudio. If my [...]

An easy way to see why that logic doesn't work is forced induction applications.  You can be at "100% load" at any manifold pressure you choose depending on your desired boost or current rpm, but that doesn't mean that 100% load should equal the barometric pressure - that would really screw stuff up.  Also, just because barometric pressure drops 20kpa doesn't mean your achievable boost target suddenly drops 20kpa.

The engine must be tuned for the actual barometric pressure it sees if using speed-density fueling.  Alpha-n is an exception because then you do need to compensate externally for changes in barometric pressure, however speed-density does not need this.  I've done testing with altitude changes of 4,000ft and more, and when using speed density you _do not_ want (or need) to use barometric correction.

I'm sure you can probably achieve a similar result the way that you are proposing, however all it's doing is adding more variables and calculations to the mix.  The fact is that your engine doesn't know the difference between 80% throttle at 95kpa or 100% throttle at 100kpa, so driving around when baro is 95kpa at full throttle isn't functionally different than driving around at 80% throttle when baro is 100kpa.  The engine is still fully loaded, but there's less air being forced in.  This is perhaps more an issue of semantics in regards to load and VE than anything.

And I do suppose that example opens up the IMAP/EMAP debate but generally speaking it's negligible for this context.  Anyway, the ideal settings for the vast majority of speed-density installs are map/100 (fixed) on the VE table and the baro correction table should be left at 100% across the board.

iltHeo84, 2023-01-20:
> And I do suppose that example opens up the IMAP/EMAP debate but generally speaking it's negligible for this context. Anyway, the ideal settings for the vast majority of speed-density installs are map/100 (fixed) on the VE table and the baro correction table should be left at 100% across the board.

I will try selecting MAP/100 (and scaling the whole VE by .97 since that was the baro value at which I tuned the most of it) and see if it works.
Going up in altitude made my engine run very rich, but at this point I assume it was due to the MAP/Baro multiply option.

closed 2023-01-26 by iltHeo84