speeduino/speeduino #990: Dropbear SD logging problem

tefnet77, 2023-01-23:
I´ve a problem with sd logging in drop beard, when on-board logging RPM is enabled the motor won’t start. If I disable it the car starts normally.
This is my tune.
[CurrentTune.zip](https://github.com/noisymime/speeduino/files/10482922/CurrentTune.zip)

noisymime, 2023-01-24:
Is it only RPM based logging that causes the problems? If you use any of the other triggers or start the car and then begin a manual SD log, is everything OK? 

Interestingly, you've got the OFF RPM to be 14300, which is somehow above the max permitted value of 10,000. I don't think that's causing a problem, but it's strange to see.

tefnet77, 2023-01-24:
Tested with manual sd logging and the same happens, the motor stops the moment you start to log.
With on boot log there is no problem and I have made a few logs with it.
Haven’t tested engine protection and external switch.
With the RPM log I try setting RPM to 2000 and the car starts but when it reaches 2000 RPM the motor stops.
The OFF RPM of 14300 I think is from the base tune but I’m not sure, tried setting to 7000 and the same happens.

noisymime, 2023-01-24:
Can you provide a log of it occurring with the RPM trigger set as you have (Starting the log at 2000 RPM)? I'm not seeing the same issue here, so a log would help see what's happening there.

tefnet77, 2023-01-24:
This is the log with the rpm set to 2000
[2023-01-24_23.02.18.mlg.zip](https://github.com/noisymime/speeduino/files/10494481/2023-01-24_23.02.18.mlg.zip)

To be honest I think I had the car running with rpm logging on, the first time I was playing with idle or cranking settings and the car stopped to start, after trying to find the cause and spending a lot of time I desisted and started a tune from scratch. 
In this second tune happened the same, after playing with idle or cranking settings the car refused to start. This time after trying a few changes I disabled the rpm logging and the car started again. So, I thought it was a logging problem but may be is something more complex than only the logging part.

adbancroft, 2026-09-06:
Issue is too old to troubleshoot- please reopen if still a problem

closed 2026-09-06 by adbancroft