speeduino/speeduino #995: 202207.01 drops comms on large burns

runesmdk, 2023-01-30:
Ch340 boards, drops comms on large burns like loading a tune or even just a full ign table. 

Both cabled and bluetooth

noisymime, 2023-01-30:
I think we might be heading towards needing to add a compatibility mode option or similar for these CH340 boards. With some quick playing around, they work fine if I slow the receive/burn rate down further, but that is a negative experience for most users who are on a 16u2 based board. 

I'll keep playing to see if I can find any other ways of getting them stable without impacting speed too much, but otherwise the compatibility option might be the best way.

runesmdk, 2023-01-31:
Any way i can implement this myself for a field fix now?

noisymime, 2023-02-01:
> Any way i can implement this myself for a field fix now?

Try changing this line:
https://github.com/noisymime/speeduino/blob/master/speeduino/storage.cpp#L163

To some lower values. The more you lower it, the longer the table writes will take, but you should fine a value around 15 that works. 

If you're seeing the saying issue with the engine running, try lowering the `15000U` figure on line https://github.com/noisymime/speeduino/blob/master/speeduino/storage.cpp#L172

runesmdk, 2023-02-01:
> > Any way i can implement this myself for a field fix now?
> 
> Try changing the `EEPROM_MAX_WRITE_BLOCK` values in `storage.cpp`:
> 
> https://github.com/noisymime/speeduino/blob/84cc382b17df65ab189269ac27554305edc9e7c6/speeduino/storage.cpp#L163-L176
> 
> The more you lower it, the longer the table writes will take, but you should fine a value around 15 that works.
> 
> If you're seeing the saying issue with the engine running, try lowering the `15000U` figure slightly further down also

Will try

noisymime, 2023-04-24:
The current master firmware has had a compatibility mode option added to the project properties that should resolve this. 

Could you please test and see if you are still having the same issue with that enabled?

runesmdk, 2023-05-26:
> The current master firmware has had a compatibility mode option added to the project properties that should resolve this.
> 
> Could you please test and see if you are still having the same issue with that enabled?

Just updated to 202207.2 - where's the compability mode hidden? There's only "New comms" under project properties now.

noisymime, 2023-10-09:
This is not in the LTS release, you can try it in either of the newer feature releases (`202305` or `202310`).

I've been testing on a CH340 board recently and am not seeing any issues with large burns in either regular or compatibility mode, so I think this one should be right to close now. 

FWIW though, these do seem to benefit from the data read speed being limited to 20/s rather than the 25/s default.

noisymime, 2023-11-11:
Closing this for now as I believe the issue is resolved in recent firmwares (202305+).

closed 2023-11-11 by noisymime